inonoob/pirowflo #3: Serial not opening

JensJN, 2021-01-12:
When attempting to connect to the WR via S4 USB (python3 waterrowerthreads.py -i s4 -b), the following error message appears:
serial open error waiting
I'm attempting to run this on an Intel NUC PC running Ubuntu 20.10.
Any advice appreciated!

inonoob, 2021-01-12:
Hey 

If you are not using a Raspberry pi you probably have right access issues. This is due that the Waterrower is connect via a serial to usb cable. In order to access that connection you must be part of the group "dialout"

Let's check if that's the case but first check if the Waterrower is seen by the system:

Check in the dev folder for "ttyACM0"

`cd /dev/ | ls -lha` 

If you see the ttyACM0 and next dialout dialout then you know the Waterrower is there. 

Let's now check if your $USER is part of that group:

`groups`

You should see in a list. if "dialout" is not there then we know you are missing to that group. To add your user to this do the following:

`sudo adduser $USER dialout`

And then reboot. 

Now it should work.

JensJN, 2021-01-12:
Hey, thanks for the quick help, much appreciated!

The above did the trick in terms of opening the serial and I'm getting all the way to:
2021-01-12 19:13:15,845 - wrtobleant - INFO - Waterrower Ready and sending data to BLE and ANT Thread

I can also see the bluetooth device/rower on Coxswain and WR Connect (on Android), but can't connect on either unfortunately. Just getting a red X on WR Connect when selecting "Yes connect to my WR" and on Coxswain "bluetooth rower connection lost to [...]". Any ideas? Thanks again for your help and let me know if you prefer I open a separate issue for this.

inonoob, 2021-01-12:
Hey again, 

yeah that's also a problem. Here the issue is with coxswain. Sven the maker of the app had to update part of his code for better Bluetooth connectivity. The issue here is that the new version wasn't accept by google on the playstore. Something went wrong. 

the solution: 
- you wait for the Coxswain app version 7.9 or 8.1 to be released on google playstore or: 

or 

- go to the https://github.com/svenmeier/coxswain/releases 
- download the app-release.apk 
- install it on you android device ( Remember to switch back install from unknown source ) 

concerning the WR Connect, I will test it. 

Best regards

JensJN, 2021-01-12:
Makes sense. I ultimately want to get this to work with Holodia/Holofit via Samsung Gear VR/Quest 2 actually (don't ask - I'm just keen to try!). Any other ideas how I might be able to test whether the bluetooth connection works in principle (other than via Coxswain or WR Connect)? I'll be able to test on Holodia itself tomorrow.

inonoob, 2021-01-12:
yes you can you could download the app nFR 

https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp 

This will show you all the bluetooth near you. If you choose the S4 COMMS PI and click connect. You will see a 5 running so called service. The one which is important is Fitness Machine. 
If you click on it you should see the characteristic "Rower Data"  click the arrows next to it. It will then ask the rower to send every 0.5 sec the data from the Waterrower. If you pull now and do some stroke you should see the numbers changing. 
With this you know it is working and you get all the needed data. 

best regards

inonoob, 2021-01-12:
Problem fixed

closed 2021-01-12 by inonoob