inonoob/pirowflo #7: add simple web interface to start/stop the application

flynch, 2021-01-24:
add a simple web interface that starts, stops and resets the application.

inonoob, 2021-01-24:
If someone want to do it please try using tornado for the web application. Avoid Django as it is super heavy for such a lightweight project.

flynch, 2021-01-24:
I had started prototyping something using flask (as I had used it once before), it's pretty lightweight and runs just fine on a pi zero. Are you okay with this or should I rework it with tornado - I would likely have something sooner with flask.

inonoob, 2021-01-24:
You might wanna check this article. https://thehftguy.com/2020/10/27/my-experience-in-production-with-flask-bottle-tornado-and-twisted/

I was also trying to do flask but this article kind of scared me of. 

I see that tornado has some advantage for i.o stuff.

flynch, 2021-01-24:
I came across http://supervisord.org/ (source over at: https://github.com/Supervisor/supervisor) which is a process management framework written in python that can start / stop processes and has a simple web interface. I just tried it out on my pizero and it appears to be very lightweight and works pretty well.
This appears to be an actively maintained project that does most of what we need (at least for this issue), so I would be inclined to use this instead of rolling our own solution. 
I need to do some more testing (in a fresh environment), to ensure my updates to the install script are correct, then I will send along a pull request that includes updates to our install script, a supervisor config file and a script to kick it off.

inonoob, 2021-01-26:
Hey 

what about that https://github.com/ateska/ramona

flynch, 2021-01-28:
fyi - I submitted a pull request that adds supervisord (https://github.com/inonoob/pirowflo/pull/8), I don't seem to have permissions to link the pull request to this issue.

inonoob, 2021-01-28:
I saw it, I need to test it.

flynch, 2021-02-13:
pull request has been merged.

closed 2021-02-13 by flynch