inonoob/pirowflo #9: use a single push button to use PiRowFlo

inonoob, 2021-01-28:
Idea:

Solder a push button on the Raspberry pi's GPIO.  if the button is push "on", a new python script should do the following: 

Pseudo code: 

while True: 

- Check if push button is True, if yes: 

  - Check if the S4 Monitor is connect and if an Ant+ dongle is connect, if yes: 
    - then start the script with the s4 monitior as input and bluetooth and ANT+ as output  `waterrowerthreads -i s4 -b -a`
    - else if start the script with the s4 monitior as input and only the bluetooth as output  `waterrowerthreads -i s4 -b`
   - if no S4 Monitor is connected, check if 2 bluetooth dongle are present and also an ANT+ dongle, if yes: 
     - then start the script with the smartrow as input and bluetooth and ANT+ as output `waterrowerthreads -i sr -b -a`
     - else if start the script with the smartrow as input and only the bluetooth as output  `waterrowerthreads -i s4 -b`
- if push button False, 
  - then sleep for 1sec 

This code should run in a separate script. which then invokes the pirowflo script

flynch, 2021-01-28:
this is a great idea. Though I suspect that relatively few people will go as far as soldering a button into place.  You don't know how many times I've wished that the raspberry pi had a few buttons. Heck, it would be nice if someone built a raspberry pi case with a header that attaches a couple of buttons (but I've yet to see one to my liking).

Maybe we should also consider having a configuration file instead or in addition to the command line options? If we do that we could add another script that inspects the peripherals that you have connected and generates a config file. We could then allow users to run this script via the web interface or maybe even just have it run when the pi boots up.
What do you think?

inonoob, 2021-01-28:
Hey @flynch 

I just order  a HAT board with an OLED and the TFT version with buttons. 
[Waveshare TFT/OLED](https://www.waveshare.com/1.44inch-lcd-hat.htm )

So we don't need to sold anything and we can use the joystick and the 3 buttons for the project. Additionally the TFT/OLED can be used to display information and status.

flynch, 2021-02-03:
I think I have one of those hats - the issue that I had with them is that I couldn't find a good case - or 3d print for a case that had reasonable buttons. 

I like the pi zeroU and the case that c4labs make for it - https://www.amazon.com/C4Labs-Zebra-Zero4U-Case-Raspberry/dp/B07V2KC86N/ref=sr_1_1

let me know if you come across a good case that works with the display / buttons

inonoob, 2021-02-05:
First test with new lib looks promising for the oled display. The button function must be done separately.  
https://github.com/rm-hull/luma.oled  

Testing for the TFT version pending. But this lib might be a good choose for it. https://github.com/juj/fbcp-ili9341

inonoob, 2021-02-22:
Close as with release v1.1.0 the 1.3inch Oled is available with buttons

closed 2021-02-22 by inonoob