inonoob/pirowflo #15: iOS keeps asking for pairing loop

inonoob, 2021-02-09:
On iOS device when trying to use pirowflo, the iOS device keeps asking to pair with pirowflo. But as it is a Bluetooth LE device it shouldn't ask for pairing with it. 
Bug must be investigate. 
[image]

```
RequestAuthorization (/org/bluez/hci0/dev_XX_XX_XX_XX_XX)
INFO:ble:RequestAuthorization (/org/bluez/hci0/dev_XX_XX_XX_XX_XX) Authorize? (yes/no): 
```

Even if accepted it keeps asking on the iOS device and on the pirowflo

flynch, 2021-02-09:
I did some further testing of this today and noticed the following: 
The issue does not occur when I run pirowflo taking its input from the s4 module and only use the built in bluetooth adapter. i.e. I do not see the issue when my usb bluetooth dongle is disconnected. 
However as soon as I connect the usb bluetooth dongle the issue occurs, even if I'm still running pirowflo in s4 mode. 
I also noticed that the order of the bluetooth adapters on my pi zero is a little surprising - when I boot up with the usb adapter attached it gets listed as hci0 while the builtin adapter gets listed as hci1. Looking at the code in waterrowerble.py it appears that it picks the first available bluetooth adapter, which made me wonder if this could this be a source of trouble?

inonoob, 2021-02-09:
Hey @flynch, 

very nice that you continued to debug. 
The order of the Bluetooth modules are as follows as the Bluetooth dongle is seen first before the script responsible for the internal Bluetooth starts. 

The SmartRow should always take the internal one "hci1" and the USB dongle is "hci0" the Broadcaster. 

We could change that the gatt server only takes a defined device e.g hci0. 

When you write that as soon as you connect the bluetooth dongle, the iOS asks to pair ?

inonoob, 2021-02-09:
Maybe the same hostname makes trouble ?

flynch, 2021-02-09:
I'm charging up my old android phone and will test with it tonight or tomorrow. I am curious as to whether there is an issue with my bluetooth dongle or if this is an ios issue? 
I will also have another bluetooth dongle to test with in a few days time.

inonoob, 2021-02-10:
Hey no it's iOS related. David from the Facebook group has the same issue with iOS

inonoob, 2021-02-10:
Hey @flynch, 

might you test the following on your running pi zero : 

sudo hciconfig hci1 sspmode 1
sudo hciconfig hci0 sspmode 1 

this will set the bluetooth modules to auto pairing. 

Let's see if that works

inonoob, 2021-02-10:
In the script ble.py 

https://github.com/inonoob/pirowflo/blob/54969702ef2e73bc530469c97de98ebd2fb9158e/src/ble.py#L385

if we replace:

```
    @dbus.service.method(AGENT_INTERFACE, in_signature="os", out_signature="")
    def AuthorizeService(self, device, uuid):
        logger.info("AuthorizeService (%s, %s)" % (device, uuid))
        authorize = ask("Authorize connection (yes/no): ")
        if authorize == "yes":
            return
        raise Rejected("Connection rejected by user")
```

authorize with "yes",then we might mitigate that issue.

flynch, 2021-02-10:
I agree and I will test this shortly. 
I note that one of the commenters on https://raspberrypi.stackexchange.com/questions/50496/automatically-accept-bluetooth-pairings linked to ecodroidlink which uses this same approach (see: https://github.com/ykasidit/ecodroidlink/blob/master/edl_agent).

inonoob, 2021-02-10:
Nice catch haven't seen and understood by now you mention it !

inonoob, 2021-02-11:
That still doesn't resolve the issue. As the iOS device keeps asking for pairing. Maybe it is an idea to test a new version of bluez. 
In order to do so, bluez must be compiled from source: 

https://scribles.net/updating-bluez-on-raspberry-pi-from-5-43-to-5-50/ 

instand of using 5.50, we use 5.55 which is the latest version of Bluez.

inonoob, 2021-02-12:
@flynch has implemented the iOS specification for Bluetooth connnection: 

```
$ cat /sys/kernel/debug/bluetooth/hci0/conn_min_interval 
12
$ cat /sys/kernel/debug/bluetooth/hci0/conn_max_interval 
12
$ cat /sys/kernel/debug/bluetooth/hci0/supervision_timeout 
500
```

Source of this info: https://stackoverflow.com/questions/55189681/unable-to-maintain-ble-connection-bluez-linux-ios/55228279?fbclid=IwAR2vnpXprviGTWCYC6AC5lJIgvvVLv446Kh4Wlk5irijiC1tk-vbc2Dds8s#55228279

flynch, 2021-02-13:
Flo and I were chatting about this on messenger - one thing that we observed was that the traffic from ios was coming in on hci1, which is odd as the gatt server is on hci0. Flo switched the interfaces that are used for smartrow and the gatt server in this commit https://github.com/inonoob/pirowflo/commit/1022d1c175b43993f8f45ca868bc81bb55541eb8?fbclid=IwAR1jfsIJa4KlabhwBD_F1_NWDjf0EXBlTroVr4t8Z5lGh3lV0Sqt9q7N0MM and my iOS devices have been able to connect without any trouble.
It would be great if other folks could also sync and confirm that this issue is resolved.

inonoob, 2021-02-15:
I now got a confirmation from a second iOS user that it now works. So we can close this bug.

closed 2021-02-15 by inonoob