inonoob/pirowflo #16: [BUG] Connecting with smartrow problem

putti23, 2021-02-12:
pi@raspberrypi:~/Repo/pirowflo/src $ python3 waterrowerthreads.py -i sr -b
2021-02-12 19:53:38,711 - __main__ - INFO - Namespace(antfe=False, blue=True, interface='sr')
2021-02-12 19:53:38,712 - __main__ - INFO - S4 not selected
2021-02-12 19:53:38,712 - __main__ - INFO - inferface smartrow will be used for data input
2021-02-12 19:53:38,713 - __main__ - INFO - Waterrower Interface started
2021-02-12 19:53:38,713 - __main__ - INFO - Start BLE Advertise and BLE GATT Server
2021-02-12 19:53:38,714 - __main__ - INFO - Ant service not used
2021-02-12 19:53:38,855 - waterrowerble - INFO - Registering GATT application...
2021-02-12 19:53:38,920 - smartrowreader - INFO - starting discovery
2021-02-12 19:53:38,991 - ble - INFO - GetManagedObjects
2021-02-12 19:53:39,024 - ble - INFO - GetAll
2021-02-12 19:53:39,029 - ble - INFO - returning props
2021-02-12 19:53:39,051 - waterrowerble - INFO - GATT application registered
2021-02-12 19:53:39,072 - waterrowerble - INFO - Advertisement registered
2021-02-12 19:53:39,614 - smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
INFO:smartrowtobleant:SmartRow Ready and sending data to BLE and ANT Thread
Update Waterrower Data
2021-02-12 19:54:01,879 - ble - INFO - Default StartNotify called, returning error
INFO:ble:Default StartNotify called, returning error
2021-02-12 19:54:01,981 - waterrowerble - WARNING - no data from s4 interface
WARNING:waterrowerble:no data from s4 interface
dbus.Array([dbus.Byte(0)], signature=dbus.Signature('y'))
Fitness machine control point: dbus.Array([dbus.Byte(0)], signature=dbus.Signature('y'))
Request control
fmcp_cb activate

2021-02-12 19:54:04,654 - smartrowreader - INFO - Connection failed [f8:c1:e4:c3:14:00]: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
INFO:smartrowreader:Connection failed [f8:c1:e4:c3:14:00]: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
dbus.Array([dbus.Byte(7)], signature=dbus.Signature('y'))
Fitness machine control point: dbus.Array([dbus.Byte(7)], signature=dbus.Signature('y'))

inonoob, 2021-02-12:
Hey I see you are trying to use the SmartRow. What I can see is that the connection to the smartrow was not established and therefore you couldn't get the data to your app.

inonoob, 2021-02-12:
Please sync with the project again, as a lot of changes had happened in between. And we are also currently having issues with SmartRow and iOS.

putti23, 2021-02-12:
> Wysłano: piątek, 12 lutego 2021 20:11
> Do: inonoob/pirowflo <pirowflo@noreply.github.com>
> DW: putek@onet.pl; Author <author@noreply.github.com>

It’s work.

Thank you

inonoob, 2021-02-12:
You are welcome !

putti23, 2021-02-12:
Unfortunatelly, problem still exists ;-(
It worked only once (I didn’t change anything)
SmartRow blink blue , but pirowlo cannot connect.
I made reboot, hard reset, but nothing helped. I cloned repo again.
Any ideas ?

RP

2021-02-12 21:04:16,145 - waterrowerble - INFO - Advertisement registered
2021-02-12 21:05:19,997 - smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
INFO:smartrowtobleant:SmartRow Ready and sending data to BLE and ANT Thread
2021-02-12 21:05:45,038 - smartrowreader - INFO - Connection failed [f8:c1:e4:c3:14:00]: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
INFO:smartrowreader:Connection failed [f8:c1:e4:c3:14:00]: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

Wysłano: piątek, 12 lutego 2021 21:04

Closed #16<https://github.com/inonoob/pirowflo/issues/16>.

inonoob, 2021-02-12:
I have some question: 

Raspberry pi type: 
Bluetooth dongle type : 
sequence you do to connect to the pirowflo? 

You first pull that the smartrow gets blue 
Then you start the script 
And it should connect 

How new is your SmartRow ?

putti23, 2021-02-12:
This Sequence worked:
I disabled BT in my phone. (SmartRow app is not working)
I have to start program first
After that activate SR
And last I active bluetooth in my phone and connect to pirowflo

But when I activate SmartRow first program cannot connect to it. I have to wait till stop blinking and activate again.

inonoob, 2021-02-12:
Hey

can you update you repo. I changed the order of the bluetooth usasge. So now the bluetooth dongle connects to the SmartRow and the internal one is used of the gatt server. Maybe that helps for the connection.

putti23, 2021-02-13:
Works like a charm. ;-)
Thank you.
I would suggest to autodisconnect while speed is 0 for more than 1 min. Now it drains battery until you stop program on raspi.

closed 2021-02-13 by putti23