inonoob/pirowflo #19: [BUG] Starting problem

putti23, 2021-02-15:
After upgrade to newest version:
python3 ./Repo/pirowflo/src/waterrowerthreads.py -i sr -b
Traceback (most recent call last):
  File "./Repo/pirowflo/src/waterrowerthreads.py", line 108, in <module>
    main(args)
  File "./Repo/pirowflo/src/waterrowerthreads.py", line 39, in main
    logging.config.fileConfig('/home/pi/pirowflo/src/logging.conf', disable_existing_loggers=False)
  File "/usr/lib/python3.7/logging/config.py", line 71, in fileConfig
    formatters = _create_formatters(cp)
  File "/usr/lib/python3.7/logging/config.py", line 104, in _create_formatters
    flist = cp["formatters"]["keys"]
  File "/usr/lib/python3.7/configparser.py", line 958, in __getitem__
    raise KeyError(key)
KeyError: 'formatters'

inonoob, 2021-02-15:
Hey 

as you don't use the default setup you path is different. I need to fix this in the following part of the code: 

https://github.com/inonoob/pirowflo/blob/69850a5f5866f127a6f5a459bace2c6c2d323152/src/waterrowerthreads.py#L39

    logging.config.fileConfig('/home/pi/pirowflo/src/logging.conf', disable_existing_loggers=False) 
    
Just replace the path with your path: 

/Repo/pirowflo/src/logging.conf

I will fix that so the script checks the absolut path first

putti23, 2021-02-15:
OK, Thank you

inonoob, 2021-02-15:
Fixed

closed 2021-02-15 by inonoob