inonoob/pirowflo #20: [BUG] Garmin Fenix 5 compatibility

putti23, 2021-02-15:
I have got Fenix 5 and garmin usb dongle .
I have problem to add pirowflo as a sensor in my watch. It doesn't see any sensor.
Maybe pirowflo works only with with fenix 6. 
In logs i can see:
2021-02-15 18:25:47,930 - __main__ - INFO - Start Ant and start broadcast data
start search for dongle
2021-02-15 18:25:48,016 - smartrowreader - INFO - starting discovery
2021-02-15 18:25:48,067 - waterrowerble - INFO - Registering GATT application...
2021-02-15 18:25:48,187 - ble - INFO - GetManagedObjects
2021-02-15 18:25:48,209 - ble - INFO - GetAll
2021-02-15 18:25:48,222 - ble - INFO - returning props
2021-02-15 18:25:48,265 - waterrowerble - INFO - GATT application registered
2021-02-15 18:25:48,272 - waterrowerble - INFO - Advertisement registered
calibration
set Networkkey:b'\x00\xb9\xa5!\xfb\xbdr\xc3E'
create Channel
----
I checked on fenix 5 and germin edhe 830. with no success.
---
Is it Bug or works only with fenix 6 ?

inonoob, 2021-02-15:
Hey 

I got the feedback that it works for the Fenix 6 with smartrow and S4 monitor. You can check if your dongle is working with the Android app ant+ plugin sampler or ipsensorman. You need to attach the dongle to your phone via org cable. 

If you can add it to your watch then you can try ergiq to see if you can connect.

inonoob, 2021-02-16:
Do you have some news for me ?

inonoob, 2021-02-16:
I just check a bit for garmin and you must ensure that the Garmin product support the following: 

- FE-C or FE support either

If it does then you can try make it work with the watch. I have the same issue as my Vivoactive 3 has no Ant+ profile on it to interact with fitness equipments

putti23, 2021-02-16:
> Wysłano: wtorek, 16 lutego 2021 12:41
> Do: inonoob/pirowflo <pirowflo@noreply.github.com>
> DW: putek@onet.pl; Author <author@noreply.github.com>

Fenix5 doesn’t have FE-C ;-(

inonoob, 2021-02-16:
But if you still Interested, we could see if we can get the Garmin 830 running as it has the FE-C and FE support

inonoob, 2021-02-23:
Closed no feedback

closed 2021-02-23 by inonoob