inonoob/pirowflo #26: Second BLE Server as Bluetooth Passthrough for SmartRow App

inonoob, 2021-02-22:
### Problem description:
Use the SmartRow app at the same time as an third party app. 

### Possible solution: 
In order to still use the SmartRow app at the same time as a third party app, a second GATT server should be started with the third available Bluetooth module. The setup should look like the following: 

- 1st Bluetooth dongle ==> connected to SmartRow device 
- Internal Raspberry pi Bluetooth ==> connection to iOS/Android device (Fitness equipment Bluetooth profile) 
- 2nd Bluetooth dongle ==> connected to iOS/Android device (SmartRow Bluetooth profile) 

the second Bluetooth GATT server would have the same Service and characteristics as the SmartRow device. The SmartRow app would send the heart beat and also reset commands to the GATT Bluetooth server which would forward those signal to the real SmartRow. At the same time the GATT Bluetooth server would send the SmartRow data to the SmartRow app. 

### Possible alternatives solution: 
Have two PI Zero send the data from one pi to the other and vis vera via netcat, if two GATT server at the same time is not possible.

inonoob, 2021-02-22:
The Gatt server for the SmartRow passthrough is running but the SmartRow apps doesn't want to connect. 
The SmartRow app sees the raspberry pi SmartRow. If clicked on connect. The app says establishing connection but doesn't pass it. 
I can't see that the app tries to send any command and I can't see that the app request notification from the smartrow. 
So there still must be a handshake beside the heart beat and reset.

inonoob, 2021-03-02:
I tried to continue with this implementation. I still hit a wall. I did find out that if I clone the PiRowFlo advertiser in the nRF app and let the SmartRow app try to connect to it, it works. The SmartRow acknowledge that the connection is working. If I try the same with only the Pi then I can't get pass the "establishing connection". Something is the app nRF doing when faking the SmartRow that the GATT bluez implementation doesn't. 

I tried some of those [link](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt) but I can't get it connect. 

But the thing is if I use the nRF app, I can connect to the PI and get the fake data from the Pi. So the App SmarRow expects something to pass the "establishing connection". I already set the recommended min max connection intervals according to the SmartRow itself. 

It might be necessary to further investigate the connection between the App and the fake SmartRow with the nRF app. And check why it is so different from the PI connection.

inonoob, 2021-03-03:
Oki, next step has been taken, with the help of the nRF app the SmartRow (PI) is bond to the smartphone and then tried to pair it within the SmartRow app then it is working. The key also is to have it bond with the smartphone first.

inonoob, 2021-03-03:
In order to work in the ble2.py the following must be added to the advertiser: 

```
        self.discoverable = None
```        

```
        if self.discoverable is not None:
            properties["Discoverable"] = dbus.Boolean(self.discoverable)
```

And in the smartrowble gatt server config the following must be add to the advertiser part: 

```
self.discoverable = True
``` 

check the api doc [link](https://kernel.googlesource.com/pub/scm/bluetooth/bluez/+/5.50/doc/advertising-api.txt) 

I needed to add it to the gatt example as it was missing and not implemented.

inonoob, 2021-03-04:
New feedback even now smartrow apps talk to PiRowFlo, I still have the issue that once the dual Gatt server has been started and then stopped and the dual Gatt server is started again dbus will complain that there is already a Gatt server running. 
I still have no idea how to avoid this error or clean dbus from left behind garbage entrie. 

Even restarting bluetooth.service don't help. The only solution I found either restart the pi or sometime if I wait long enough then it seems to work again without reboot. 

Will post error Code later

inonoob, 2021-03-04:
Oki this thing is driving me CRAZY !!! Now it is not working anymore ....This SmartRow app won't connect to a fresh install raspbian with pirowflo !!! Damnn!!

inonoob, 2021-03-05:
Oki I got it this time ! 
It was because the Raspberry pi Bluetooth was set to Bluetooth Classic and LE_only. If both are activated at the same time then SmartRow App and additionally the WaterRower Connect app won't work under Android !! They must be set to LE_only.

In order to fix that issue the following is possible:
- either set in the option under `/etc/bluetooth/main.conf`: 

```
# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
#ControllerMode = le
```

- Or to re-compile bluez and change the default value in the `main.c` of bluez: 

```

static int get_mode(const char *str)
{
	if (strcmp(str, "dual") == 0)
		return BT_MODE_DUAL;
	else if (strcmp(str, "bredr") == 0)
		return BT_MODE_BREDR;
	else if (strcmp(str, "le") == 0)
		return BT_MODE_LE;

	error("Unknown controller mode \"%s\"", str);

	return BT_MODE_DUAL;
}
```

https://github.com/bluez/bluez/blob/a37d53db9ae7d21a8f812925303d767d3f03e597/src/main.c#L306-L318

The issue with compiling bluez is that in newer version they added things like `Device Information` ==> `PnP ID`

inonoob, 2021-03-07:
Oki, 

SmartRow now can talk to PiRowFlo and the data are also send to the App it is working but. I only can have either smartrow or normal bluetooth le with the Fitness Equipement profile ! 

When I try to start the second Bluetooth Gatt server in this case the SmartRow LE signal then D-Bus always quit the thread with the following error message: 

```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "waterrowerthreads.py", line 55, in BleService2
    bleService2 = smartrowerble.main(out_q, ble_in_q_sr)
  File "/home/pi/pirowflo/src/smartrowerble.py", line 268, in main
    agent_manager.RegisterAgent(AGENT_PATH, "NoInputNoOutput") # register the bluetooth agent with no input and output which should avoid asking for pairing
  File "/usr/local/lib/python3.7/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/local/lib/python3.7/dist-packages/dbus/proxies.py", line 147, in __call__
    **keywords)
  File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 653, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.AlreadyExists: Already Exists
```

I don't know how to fix this issue with the double Gatt server running at the same time !

inonoob, 2021-03-11:
Oki, 

I can't find a solution for that. The only idea would be to shift the second GATT server in a complete new process ! I reached my knowledge point with D-BUS where I can't get any further and debug why I can't have both GATT servers at the same time in different threads. 
I got a feedback from the bluez issue forum but I can't implemented, as I don't fully understand what is meant. 
[link](https://github.com/bluez/bluez/issues/103) 

If someone has more skills, please help :)

FlorentLa, 2021-04-04:
Hi,

Is it possible to have an option to either launch the SmartRow Bluetooth profile or Fitness equipment Bluetooth profile ? The use cas is connecting  both a Garmin watch through ant and the smart row app through Bluetooth?

Regards,
Florent

dbffm, 2021-05-11:
Hi all, 

I have the same use case then FlorentLa

> The use case is connecting both a Garmin watch through ant and the smart row app through Bluetooth

So for me there is no need to connect to coxswain or another app. I would like to go with SmartRow and take the advantage of receiving all the data to my garmin FR 745 via Ant+

> SmartRow now can talk to PiRowFlo and the data are also send to the App it is working but. I only can have either smartrow or normal bluetooth le with the Fitness Equipement profile !

@inonoob Can you tell me in which script this has been implemented? I found fakesmartrowthreads.py in folder /src/testing but I assume that this is not the one I am searching for.

hnfhtw, 2021-12-09:
Hi all, any news on that topic? I have the same problem: I want to use the original SmartRow app on my Android phone and at the same time I want to see the SmartRow data in Garmin Connect (I use a Fenix 6x Pro watch). 

As far as I have tried it, the SmartRow can only connect either to the phone (SmartRow app) or to the Raspberry PI (pirowflo project) but not to both. So when it it possible that the Raspberry PI forwards the SmartRow data 1:1 to the Android phone (SmartRow app) it would be great.

@inonoob First of all thank you for that great project! Could you please give as some thoughts where and what needs to be implemented to enable that SmartRow data forwarding? Do you think it is even possible? I am not into BLE a lot but I could imagine that the SmartRow app can detect if the data was sent from the SmartRow or from another device such as the Raspberry?

inonoob, 2022-01-07:
Hey, 

So the thing is in order to it. It would require to copy the smartrow Bluetooth protocol as they have defined their own profile. In addition the Ant+ protocol must work with the data. 

I haven't implemented. I wanted to start but I had issues with Bluetooth. I wanted to have 1 Bluetooth input smarter then as output 2 Bluetooth. The first would be the Bluetooth with fitness rowing profile and the second would be the relay for smartrow app. But as I mentioned, I did hit a wall with the two gatt server running on the same Device. As I don't have a Fenix 6 I couldn't test the native rowing app.

dan06, 2022-02-11:
@inonoob, have you written any code for the SmartRow pass-through?  Or do you have documentation on their interface?   If the issue is simply running the two BLE servers in separate processes, that should be easy to sort out with shared memory or pipes or some such inter process communication.  

I have built up a little system with the display, Pi Zero W, USB / ETH hat and extra BT dongle.  I got sample (fake) data to my Garmin FR945 using your code.  I am waiting for my SmartRow to be shipped (they are taking *forever*).  As soon as I get it, I will start playing with the code.  If you have already written some code for the pass-through, I am willing to give it a shot.  At first I won't need two BLE servers, since I don't care about BLE-FE.  I'll stick to ANT-FE to the Garmin.

dan06, 2022-03-19:
@inonoob can you share the code where you have the raspberry pi sending data to the smart row app? I've been working on this using the rest code in the repository but I can't get data to flow. I can get a connection but when I click start on the app, it never starts to show data. I just get a spinner.

I think there is a good number of people that will be happy with a solution where we use ant and ble passthrough only, which does not require 2 gatt servers

dan06, 2022-03-25:
I have the basics going to pass SmartRow data to the SmartRow app.  This does not support the BLE fitness profile at the same time.  It requires two BlueTooth devices.  I use a Raspberry Pi Zero W and one external dongle.  It will read data from the SmartRow power meter and transmit via ANT+ and the "fake" SmartRow that the app can connect to.  I have a SmartRow V3 and you will probably need the latest version of the app.

I still need to do a lot of testing to see how robust it is.  At the moment, you need to start the program via a console command line.  I have only been testing on Android.  If you are interested in helping to test this, please let me know.  I may be ready to check something into my fork over the weekend.  I will see how easy it is to integrate this with the SuperVisor console.  Or maybe editing a config file once.  Also, if you want to test, make a backup of your original piroflow directory.

Below is a short workout that I recorded on both my Garmin 945 and the SmartRow app.  SR App is left, GC is right.

Distance and stroke counts are identical.  Average power is about 10W lower on the Garmin (87W vs 98W), but that may not mean anything (counting zeros, etc).  The Garmin Normalized Power (97W) is pretty close to the SmartRow app.  I don't have a SR account, so not sure if it provides more data for comparison.

[image]

cc: @hnfhtw, @dbffm, @FlorentLa

dan06, 2022-04-22:
I've checked in a release candidate to my fork if anyone wants to test this.  No extra configuration is necessary.  I tested on a Pi Zero W with an external Bluetooth dongle with the app running on Android.  

The system must be configured for input from SmartRow and output to ANT+.  It will detect this and then automatically enable SmartRow passthrough.

1.  Configure the system for input from SmartRow and output to ANT+ (-i sr -a).
2. DO NOT start the phone app yet.
3. Pull the handle and make sure the SmartRow is connected (blue lights will go off).
4. Now start the phone app.  If it doesn't connect automatically, then touch the Bluetooth icon and select the SmartRow-44 device.
5. It should connect.
6. You should be able to do any workout on the app, and also record on your Garmin / ANT+ device.

Please let me know if you've tried it and if it worked or not.  If not, please attach piroflow.log.

dan06, 2022-04-24:
I committed two changes after doing a rowing workout today.  The first was to include the stroke count in the fields that are reset whenever the app requests a reset (the other two are time and distance). 

I recorded an hour session on my Garmin.  During this time, I did two 30 minute workouts from the SmartRow app.  I noticed the stroke count discrepancy when I compared the workouts.  Other than that, it was pretty cool, and exactly what I wanted.

The second fix was to be more deterministic when choosing Bluetooth adapters.  I coded the system for the real SmartRow to connect to the on-board Bluetooth dongle.  The fake SmartRow advertises on the external dongle.  If the external  USB dongle is plugged in when the Pi boots, then it gets hci0 and the internal dongle gets hci1.  When that happens, the real SmartRow won't connect to PiRowFlow.  Not sure why - I don't *think* I changed any of that code.  I have now changed the code that, only if passthrough is enabled, the preferred Bluetooth adapters will be selected.  This seems to yield a much more predictable system.

@inonoob let me know if you're interested in me making a pull request for this.  I am somewhat wary of breaking existing code.  I tried hard to write the code such that it only impacts the pass-through functionality.  There may be an issue with the new adapter selecting code if there are no on-board Bluetooth adapters - not sure if that is still a "thing".

Motrail, 2022-08-28:
@dan06 Hi Dan - Sorry to bother you, but I'm really interested in using this, is there any way I can do this? And do you know, if @inonoob is still developing this project? Been quiet for some time...

dan06, 2022-08-30:
@Motrail do you already have PiRoFlow working from @inonoob's branch?  If so, you should be able to install from my fork and get the pass through software.  There is no additional configuration needed, other than you need two Bluetooth devices.  I use a Pi ZeroW, with one external Bluetooth dongle.

Just to be clear, my code has only been tested for SmartRow -> PiRowFLow -> Android SmartRow App and simultaneously to a Garmin FR945.

FYI SmartRow has just updated their app to do passthrough from their app to a few other apps.  So this would not require any other hardware if you want to use the SmartRow with the few other apps they support (Kinomap and Zwift and some other app).  They are re-broadcasting the SmartRow data as Bluetooth FTMS.  This is absolutely a step in the right direction and has the potential to make this project obsolete, at least if you have a SmartRow.  However, I have not been able to get the new SmartRow app to work with my Garmin watch and a 3rd party ConnectIQ app that supports FTMS.  I have also not tested the new app with my PiRoFlow setup.  Hopefully they didn't make changes that breaks that part.

If you try this, please let me know how it goes.  I'd be happy to try and help.

Yann-F, 2022-08-30:
Hi,
@dan06.
First of all, than for your works and thanks for sharing it :)
I've installed your version of PiRowFLow from your repository on a raspberry Pi 3B+.
Then I launch waterrowerthreads.py -i sr -a
The BT connection SmartRow -> PiRowFLow works (blue light go off)
The BT connection iRowFLow -> Android SmartRow App works too, I see datas into SmartRow app.
The next step for me should connecting PiRowFLow ANT+ to my garmin Venu and the native rower gamin's app
But I have no idea to peer PiRowFLow and the garmin watch.
On the garmin watch I've done a search for new sensor but nothing was found.
Could you give me a clue to do it ?

Thanks a lot.

dan06, 2022-08-31:
@Yann-F I'm afraid I don't think the Venu supports the ANT+ profile for this feature.  This also does not work on my wife's Vivoactive.  You'll need a watch that supports the FE-C (Fitness Equipment) profile.  This is only available on Garmin's higher end watches.

You can check https://www.thisisant.com/directory/ and search for your device.

Yann-F, 2022-08-31:
Ok that why It doesn't work with my watch.
Thanks for the info @dan06.

stampede79, 2022-09-28:
Hi @dan06 
Thx for your hard work.
Unfortunatly I can't get your code working on my pi zero w.
When I use the internal bluetooth adapter I am able to start waterrowerthreads.py -i sr -a, and if smartrow is able to connect (thats not always possible) my garmin wath (fenix 7) gets no data at all.
When I disable internal bluetooth an use a dongle, waterrowerthreads.py -i sr -a does not launch anymore. 
2022-09-28 11:45:17,050 - __main__ - INFO - Start Ant and start broadcast data
start search for dongle
Can't get device info: No such device
Can't get device info: No such device 

perhaps you are able to help. Thx in advance,

stampede79, 2022-09-28:
Hi @dan06 
first of all thx for your effort.
Your version is now running on my raspi. But the SR-App won't connect to PiRowFlo, even it appears sometimes for a second in the available devices "SmartRow-44". I got the following heartbeat "error" in the log:
` INFO - 	[e8:f6:a2:1e:c4:69] Service [00001801-0000-1000-8000-00805f9b34fb]
2022-09-28 14:01:02,154 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
starting heart beat
invalid literal for int() with base 10: '000\rS'
a000
SmartRow0'V201+
2022-09-28 14:01:06,214 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2022-09-28 14:01:06,409 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using /org/bluez/hci0
2022-09-28 14:01:06,478 - adapters.fakesmartrow.fakesmartrowble - INFO - Registering GATT application...
2022-09-28 14:01:06,532 - adapters.fakesmartrow.ble - INFO - GetManagedObjects
2022-09-28 14:01:06,549 - adapters.fakesmartrow.ble - INFO - GetAll
2022-09-28 14:01:06,565 - adapters.fakesmartrow.ble - INFO - returning props
2022-09-28 14:01:06,586 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2022-09-28 14:01:06,602 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered`  

Perhaps you or someone else may help me.
Thx in advance.

dan06, 2022-09-28:
@stampede79 can you attach the full log?  

Does the SmartRow power meter stay connected to the Pi?  I.e. blue lights go off and stay off?

What device do you use to run the SmartRow app?  I have only tested on Android.

Are you getting the info via ANT+ to a Garmin or such?

stampede79, 2022-09-29:
Hi dan06,

thx for your quick answer.
I use an android phone (Samsung) for the SR-Ap. Tried an Fire HD Tablet as well without success.
I get the data on my garmin device (Fenix 7) without any problems.

Sometimes the rower appears in the SR App (Smartrow-44). The app says "Establishing connection...) but never returns.
The blue light at the rower turns and stays off.

content of /src/piroflow.log:
`2022-09-29 11:25:03,336 - __main__ - INFO - S4 not selected
2022-09-29 11:25:03,346 - __main__ - INFO - interface smartrow will be used for data input
2022-09-29 11:25:03,355 - __main__ - INFO - Smartrow Interface started
2022-09-29 11:25:03,356 - __main__ - INFO - SmartRow passthrough is enabled
2022-09-29 11:25:03,382 - __main__ - INFO - Start SmartRow Passthrough BLE Advertise and BLE GATT Server
2022-09-29 11:25:03,389 - __main__ - INFO - Bluetooth service not used
2022-09-29 11:25:03,405 - adapters.fakesmartrow.fakesmartrowble - INFO - Waiting for real SmartRow to connect
2022-09-29 11:25:03,421 - __main__ - INFO - Start Ant and start broadcast data
2022-09-29 11:25:03,520 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2022-09-29 11:25:03,695 - adapters.smartrow.smartrowreader - INFO - starting discovery
2022-09-29 11:25:09,370 - root - INFO - found SmartRow
2022-09-29 11:25:09,380 - root - INFO - e8:f6:a2:1e:c4:69
2022-09-29 11:25:09,398 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2022-09-29 11:25:09,489 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2022-09-29 11:25:15,510 - adapters.smartrow.smartrowreader - INFO - Connected to [e8:f6:a2:1e:c4:69]
2022-09-29 11:25:16,218 - adapters.smartrow.smartrowreader - INFO - Resolved services [e8:f6:a2:1e:c4:69]
2022-09-29 11:25:16,224 - adapters.smartrow.smartrowreader - INFO -     [e8:f6:a2:1e:c4:69] Service [00001234-0000-1000-8000-00805f9b34fb]
2022-09-29 11:25:16,230 - adapters.smartrow.smartrowreader - INFO -             Characteristic [00001236-0000-1000-8000-00805f9b34fb]
2022-09-29 11:25:16,236 - adapters.smartrow.smartrowreader - INFO -             Characteristic [00001235-0000-1000-8000-00805f9b34fb]
2022-09-29 11:25:16,241 - adapters.smartrow.smartrowreader - INFO -     [e8:f6:a2:1e:c4:69] Service [00001801-0000-1000-8000-00805f9b34fb]
2022-09-29 11:25:16,301 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
2022-09-29 11:25:20,362 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2022-09-29 11:25:20,530 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using /org/bluez/hci0
2022-09-29 11:25:20,651 - adapters.fakesmartrow.fakesmartrowble - INFO - Registering GATT application...
2022-09-29 11:25:20,709 - adapters.fakesmartrow.ble - INFO - GetManagedObjects
2022-09-29 11:25:20,726 - adapters.fakesmartrow.ble - INFO - GetAll
2022-09-29 11:25:20,741 - adapters.fakesmartrow.ble - INFO - returning props
2022-09-29 11:25:20,754 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2022-09-29 11:25:20,764 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered`

content of supervisor WebUI log:
` INFO - 	[e8:f6:a2:1e:c4:69] Service [00001801-0000-1000-8000-00805f9b34fb]
2022-09-29 11:25:16,301 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
starting heart beat
invalid literal for int() with base 10: '000@\r'
a000@
SmartRow0'V201
2022-09-29 11:25:20,362 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2022-09-29 11:25:20,530 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using /org/bluez/hci0
2022-09-29 11:25:20,651 - adapters.fakesmartrow.fakesmartrowble - INFO - Registering GATT application...
2022-09-29 11:25:20,709 - adapters.fakesmartrow.ble - INFO - GetManagedObjects
2022-09-29 11:25:20,726 - adapters.fakesmartrow.ble - INFO - GetAll
2022-09-29 11:25:20,741 - adapters.fakesmartrow.ble - INFO - returning props
2022-09-29 11:25:20,754 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2022-09-29 11:25:20,764 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered`

In case you need further information, please contact me. Thx so much in advance!

Regards Jan

dan06, 2022-09-29:
@stampede79 I use a Samsung S8+ with the latest version of the SR app that works for me.  I have a V3 SmartRow, whereas you seem to have a V2.  I'll take a look at the code and see if something jumps out. 

Are you familiar enough with the system to run piroflow from the command line in a terminal emulator like Putty?  The program prints out quite a bit more info on the command line.

I suspect the issue is that the code does not properly emulate a V3 device when a V2 device is connected to it.  If you see the SmartRow-44 device at least that means the BT advertising is working.  What you are describing is what I experienced when I developed the code.  I would see the device but it would never finish connecting.  The SR app is quite picky and doesn't connect if the data isn't correct.

stampede79, 2022-09-29:
@dan06
here is the log running pirowflo from the terminal until I try to connect with SR-App. No other garmin devices are connected.

`2022-09-29 14:16:21,366 - __main__ - INFO - S4 not selected
2022-09-29 14:16:21,377 - __main__ - INFO - interface smartrow will be used for data input
2022-09-29 14:16:21,386 - __main__ - INFO - Smartrow Interface started
2022-09-29 14:16:21,387 - __main__ - INFO - SmartRow passthrough is enabled
2022-09-29 14:16:21,412 - __main__ - INFO - Start SmartRow Passthrough BLE Advertise and BLE GATT Server
2022-09-29 14:16:21,419 - __main__ - INFO - Bluetooth service not used
2022-09-29 14:16:21,433 - adapters.fakesmartrow.fakesmartrowble - INFO - Waiting for real SmartRow to connect
2022-09-29 14:16:21,448 - __main__ - INFO - Start Ant and start broadcast data
start search for dongle
2022-09-29 14:16:21,556 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2022-09-29 14:16:21,761 - adapters.smartrow.smartrowreader - INFO - starting discovery
Using Dynastream Innovations dongle
calibration
set Networkkey:b'\x00\xb9\xa5!\xfb\xbdr\xc3E'
create Channel
2022-09-29 14:16:58,786 - root - INFO - found SmartRow
2022-09-29 14:16:58,790 - root - INFO - e8:f6:a2:1e:c4:69
2022-09-29 14:16:58,802 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2022-09-29 14:16:58,872 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2022-09-29 14:17:07,024 - adapters.smartrow.smartrowreader - INFO - Connected to [e8:f6:a2:1e:c4:69]
2022-09-29 14:17:07,781 - adapters.smartrow.smartrowreader - INFO - Resolved services [e8:f6:a2:1e:c4:69]
2022-09-29 14:17:07,786 - adapters.smartrow.smartrowreader - INFO -     [e8:f6:a2:1e:c4:69] Service [00001234-0000-1000-8000-00805f9b34fb]
2022-09-29 14:17:07,792 - adapters.smartrow.smartrowreader - INFO -             Characteristic [00001236-0000-1000-8000-00805f9b34fb]
2022-09-29 14:17:07,797 - adapters.smartrow.smartrowreader - INFO -             Characteristic [00001235-0000-1000-8000-00805f9b34fb]
2022-09-29 14:17:07,801 - adapters.smartrow.smartrowreader - INFO -     [e8:f6:a2:1e:c4:69] Service [00001801-0000-1000-8000-00805f9b34fb]
2022-09-29 14:17:07,977 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
starting heart beat
invalid literal for int() with base 10: '00\rSm'
SmartRow0'V201+'
2022-09-29 14:17:12,035 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2022-09-29 14:17:12,312 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using /org/bluez/hci0
2022-09-29 14:17:12,376 - adapters.fakesmartrow.fakesmartrowble - INFO - Registering GATT application...
2022-09-29 14:17:12,429 - adapters.fakesmartrow.ble - INFO - GetManagedObjects
2022-09-29 14:17:12,445 - adapters.fakesmartrow.ble - INFO - GetAll
2022-09-29 14:17:12,456 - adapters.fakesmartrow.ble - INFO - returning props
2022-09-29 14:17:12,474 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2022-09-29 14:17:12,488 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered`

Perhaps its an SmartRow version issue as you mentionned. Noob-question: Is Smartrow updateable ?

dan06, 2022-09-29:
@stampede79 I don't think the SR App is completing the connection to the Pi.  I had some issues on my system.  

Check the StartNotify() function in fakesmartrowble.py.  It contains a couple of debug print statements in all caps about starting to notify.  If you don't see those, then something is not right.  You can add a print statement just before "GLib.timeout_add(50, self.Waterrower_cb)" call in StartNotify() and see if you get that printed.  I had a suspicion that that function didn't return.  But it's been a while.

It may also be possible that I have updated my Pi's software.  I know I spent a bit of time working through the Bluetooth connection issues.

You should get the following line in the log (bold one is missing in your case):
2022-08-30 11:59:47,021 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2022-08-30 11:59:47,031 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered
**2022-08-30 12:00:02,952 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification**

Everything else seems fine in the log, at least compared to mine.  It finds the expected BT devices.  FYI, the basic premise is that the code advertises the BLE service, which is what makes the app see the device.  It then connects, which should cause the callback to start the notify task.  That is where the data starts to stream to the app.  So in you case it looks like the notification thread is not starting. 

I am not aware if the SR device can be updated - if it could I would imagine the SR App would to it.  But I don't think this is the issue you're having.

stampede79, 2022-09-30:
@dan06 
I think i finally get it.
I did a complete reinstallation of rasberry os and pirowflo.
After restarting the service i saw, that the wrong adapters were used for the connections (SR -> Piroflow internal Bluetooth and Pirowflo -> SR-APP dongle). So I changed the def get_sr_preferred_adapter() method in src/adapters/fakesmartrow/ble.py and src/adapters/smartrow/smartrowreader.py (USB -> UART and the other way round). First test seems to run...

dan06, 2022-10-05:
@stampede79 is it working for you now?

luxmoggy, 2022-10-05:
2022-10-05 18:09:49,092 - __main__ - INFO - S4 not selected
2022-10-05 18:09:49,092 - __main__ - INFO - interface smartrow will be used for data input
2022-10-05 18:09:49,092 - __main__ - INFO - Smartrow Interface started
2022-10-05 18:09:49,093 - __main__ - INFO - SmartRow passthrough is enabled
2022-10-05 18:09:49,094 - __main__ - INFO - Start SmartRow Passthrough BLE Advertise and BLE GATT Server
2022-10-05 18:09:49,099 - __main__ - INFO - Bluetooth service not used
2022-10-05 18:09:49,100 - adapters.fakesmartrow.fakesmartrowble - INFO - Waiting for real SmartRow to connect
2022-10-05 18:09:49,102 - __main__ - INFO - Start Ant and start broadcast data
2022-10-05 18:09:49,104 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci0
2022-10-05 18:09:49,111 - adapters.smartrow.smartrowreader - INFO - starting discovery
2022-10-05 18:09:59,166 - root - INFO - found SmartRow
2022-10-05 18:09:59,168 - root - INFO - ce:69:15:7c:fc:78
2022-10-05 18:09:59,172 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2022-10-05 18:09:59,205 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci0
2022-10-05 18:10:00,942 - adapters.smartrow.smartrowreader - INFO - Connected to [ce:69:15:7c:fc:78]
2022-10-05 18:10:02,005 - adapters.smartrow.smartrowreader - INFO - Resolved services [ce:69:15:7c:fc:78]
2022-10-05 18:10:02,006 - adapters.smartrow.smartrowreader - INFO - 	[ce:69:15:7c:fc:78] Service [00001234-0000-1000-8000-00805f9b34fb]
2022-10-05 18:10:02,006 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001236-0000-1000-8000-00805f9b34fb]
2022-10-05 18:10:02,007 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001235-0000-1000-8000-00805f9b34fb]
2022-10-05 18:10:02,008 - adapters.smartrow.smartrowreader - INFO - 	[ce:69:15:7c:fc:78] Service [00001801-0000-1000-8000-00805f9b34fb]
2022-10-05 18:10:02,060 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
2022-10-05 18:10:06,074 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2022-10-05 18:10:06,074 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
Can't get device info: No such device
Can't get device info: No such device
2022-10-05 18:10:06,158 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using None
2022-10-05 18:10:06,160 - adapters.fakesmartrow.fakesmartrowble - CRITICAL - GattManager1 interface not found
Exception in thread srpt:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/iain/pirowflo/src/waterrowerthreads.py", line 81, in SmartRowPassthrough
    FakeSmartRowBLE()
TypeError: 'NoneType' object is not callable

dan06, 2022-10-05:
@luxmoggy I fixed the crash you saw.  The basic ANT+ feature should work without a second BT dongle.

I did not extensively tested it but the program stays up.  Please get the latest code and let me know if it works now.

luxmoggy, 2022-10-05:
2022-10-05 21:40:18,193 - adapters.fakesmartrow.fakesmartrowble - CRITICAL - GattManager1 interface not found
2022-10-05 21:40:18,194 - __main__ - ERROR - SmartRow passthrough exited!

A different error now. I again searched for sensors on my watch nothing (Garmin Epix 2) which should find the ANT device.

luxmoggy, 2022-10-05:
Don't spend time on this. The next time I can try is on Friday. When I hope that I will also have the BT dongle.

luxmoggy, 2022-10-05:
I am using a newish Garmin ANT+ dongle. I started reading the code, I know enough to get me in trouble. I am going to add extra logging lines to see what is happening. It must be an edge case.

dan06, 2022-10-05:
> > @stampede79 is it working for you now?
> 
> @dan06 everything works smoothly after the changes i did.

Hmmm - there is some stuff in the install.sh script that sets certain parameters on Bluetooth devices.  The issues may be related to that.  I know I messed around with some of that stuff while I was trying to get the app to connect.  I should maybe do a clean install at some point and see how that works.

It would be nice to have an install that "just works" for everybody.

Maybe starting next month I'll have some more time to play with this again.  I don't row indoors much in the summer.  I've been trying to get the battery status over ANT+ but have not been successful so far.

luxmoggy, 2022-10-06:
Looking at the code. I think that in fakesmartrowble where it end for me. Might work if at sometime a second dongle was at one time attached or like you said played with Bluetooth to get both connections from the one. The question is would people really want to get data from smartrower and just pass it on via ANT not also to the App.

dan06, 2022-10-06:
@luxmoggy fakesmartrowble thread or another BT dongle is absolutely **not** necessary to get ANT+ to work.  If your search for sensors on your Epix and not seeing it, it means the ANT+ dongle is not being initialized properly.  Also go through your list of current sensors on the Epix and make sure that it's not already paired.  If it is, remove it and re-pair it.  I wasted a lot of time thinking I don't see the device, when it was already paired on my FR945.

If you still have issues after you get the new dongle, you can start with innonoob's original code to make sure that works.

Before my changes, there was no way to send data to the SR App.  Passing it on to ANT+ allows me to have all my fitness data in the Garmin Connect ecosystem as one cohesive system.  What I like about the SR app is I can now do their structured workouts while still recording the data in Garmin Connect.

luxmoggy, 2022-10-06:
@dan06 I added extra logging comments. After this “main - ERROR - SmartRow passthrough exited!” it doesn’t run another line of code in that script.  

It is in the next few lines of code to n that script which starts the ANT+. I can see in the install it finds the dongle.

Before inserting the BT dongle I will try with innonoob’s code. To check the ant+ dongle.

luxmoggy, 2022-10-07:
So i tried inonoob's code. the code run and picks up the smart rower. but still can't find it on my watch.
From the install, so it finds the dongle. This is the first time I am using the dongle with the watch. Only needed to upgrade the firmware on the bike lights before this.

-------------------------------------------------------
check for Ant+ dongle in order to set udev rules       
Load the Ant+ dongle with FTDI driver                  
and ensure that the user pi has access to              
-------------------------------------------------------
 
No Ant stick found !
No Ant stick found !
No Ant stick found !
No Ant stick found !
Ant dongle found
udev rule written to /etc/udev/rules.d/99-garmin.rules

luxmoggy, 2022-10-07:
I have ordered another ANT+ dongle.

luxmoggy, 2022-10-08:
So with new Ant dongle. I get information to my watch. But it doesn't find the BT for the passthrough. Is says it is disabled. 

I will reinstall the pi. Keep the user as pi. As I need to change the install script due to using a different user.

How can I tell that the pi is connected to smartrow?

luxmoggy, 2022-10-08:
Now that isn't working. What OS should I install? I want to do a clean install and wonder if "Debian Bulleye with the Raspberry PI desktop" is the right starting point. Also before I do the installation is there a way to check if the dongles are there?

luxmoggy, 2022-10-08:
[image]
So I am now getting constant results. The program is working well. I get an issue with the iOS application. I am constantly asked to pair the BT - the same results if I cancel or accept - but if I accept I need to delete the BT connect or the PiRowFlo think that the BT is in use. 

When I connect to SmartRow-44  I get the "Lost Connection to the Device" straight away. It looks like it is still connected but if I click on StartRow it goes into a waiting loop, I think waiting for the passthought to talk to it.

dan06, 2022-10-11:
@luxmoggy sorry, I was out of town this weekend.  

I have not tested this with IOS. The install.sh script has a section about updating Bluetooth settings for Apple.  You can take a look there and see if that needs to be modified to change settings for the correct dongle.  I.e. the dongle that is used for passthrough.

Since I don't have any IOS devices I can't test that part.  Do you have access to an Android device that you can verify with?  FYI I don't get any pairing requests on Android.

dan06, 2023-02-03:
I posted a fix to the SmartRow passthrough code.  Doing interval workouts on the SR app didn't proceed properly from one step to the next.  I believe this has been fixed now.

Yann-F, 2023-02-11:
@dan06

Where can I found your fix about the SmartRow passthrough code ?
I use it with a Garmin Forunner 255 and yes the SR app doesn't go correctly to the next step during a interval workouts.
By the way, when I stop a record on the SR app and start a new one, the time start immediately instead of when I start rowing.
A big thanks for your work and help.

dan06, 2023-02-11:
> 
@Yann-F, the fix should be available on my fork (dan06).  Please get it again from GitHub.  I think you need to run install.sh again after getting the code. Please check the update instructions.  I may have forgotten to push the fix to GitHub.  I pushed this morning to make sure it is on GitHub.

My last commit should fix both issues you mention.  If you start a second workout in the SmartRow app, it will again wait until you start rowing again.

Please let me know if there are still issues.

Yann-F, 2023-02-11:
@dan06 
Ok now I see a new commit in your fork.
I have made a clean install on my pi 3B+ from your fork and made a quick test.
I'm still able to connect SR/pirowflo/garmin watch and SmartRow.app (on android phone)
The interval workouts now works fine and go correctly to the next step after each interval.
Starting an second workout and SmartRow.app wait until begining rowing again.

Thanks a lot for this two fixes 👍

gabrykkk, 2023-04-24:
Hi @dan06 , thanks for your fork as it's cool to keep using SR app with garmin watch sync!
I tried a clean install, but I am having some problems to connect the app, I can see SmartRower-44 and pair, but the I get the "Lost connection to the device" message. With the debug started I can see it doesn't get in the right loop as I think is always None:

```
-->c@@@@A  0   00D4
-->d@@@@A000   0-F2
-->e@@@@A000000 w5D
-->f@@@@A---00!--09
STARTING NOTIFICATION!
DONE 2 STARTING NOTIFICATION!
2023-04-24 13:48:12,062 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification
-->b@@@@A   19 13F1
2023-04-24 13:48:12,112 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: None **added by me
2023-04-24 13:48:12,113 - adapters.fakesmartrow.fakesmartrowble - WARNING - no data from SmartRow interface
2023-04-24 13:48:12,162 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: None
2023-04-24 13:48:12,163 - adapters.fakesmartrow.fakesmartrowble - WARNING - no data from SmartRow interface

 else:
            logger.info('Sending: '+str(smartRowFakeData).replace('\r', '\\r'))
            logger.warning("no data from SmartRow interface")
            pass

```
I then tried to force it as connected changing
       if (AppConnectState == AppConnectStateEnum.Connected):
to:
          if (AppConnectState != AppConnectStateEnum.Connected):

and now the connections is up and stable, I get data on both garmin and SR app.
The only problem is that it's missing all the other functions and it doesn't reset the data, disconnect etc.
Can you help me to sort out why the state is not correctly updating?
Raspberry 3 with the new SmartRow (black plastic), tried with ANT+ and BT dongle and also with 2 BT dongle.
Thanks

```
after  the change:
-->e@@@@@000000 w5C
-->f@@@@@---68!--16
-->b@@@@@   00  0D2
STARTING NOTIFICATION!
DONE 2 STARTING NOTIFICATION!
2023-04-24 13:51:24,681 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification
-->c@@@@@  0   00D3
2023-04-24 13:51:24,728 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: c@@@@@  0   00D3\r
2023-04-24 13:51:24,778 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: None
2023-04-24 13:51:24,780 - adapters.fakesmartrow.fakesmartrowble - WARNING - no data from SmartRow interface
-->d@@@@@000   0-F1
2023-04-24 13:51:24,830 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: d@@@@@000   0-F1\r
2023-04-24 13:51:24,880 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: None
2023-04-24 13:51:24,881 - adapters.fakesmartrow.fakesmartrowble - WARNING - no data from SmartRow interface

*** closing the app to try to reset

2023-04-24 13:56:57,241 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: c@@@@E  0   00D8\r
2023-04-24 13:56:57,245 - adapters.fakesmartrow.fakesmartrowble - INFO - Ending notification
2023-04-24 13:56:57,246 - adapters.fakesmartrow.fakesmartrowble - INFO - Resetting app connection
2023-04-24 13:56:57,294 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: None
2023-04-24 13:56:57,295 - adapters.fakesmartrow.fakesmartrowble - WARNING - no data from SmartRow interface
-->d@@@@E000   0-F6
-->e@@@@E000000 w61
***

-->b@@@@E  179 2812
-->c@@@@E  0   00D8
STARTING NOTIFICATION!
DONE 2 STARTING NOTIFICATION!
2023-04-24 13:57:01,762 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification
2023-04-24 13:57:01,812 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: c@@@@@  0   00D3\r
-->d@@@@E000   0-F6
2023-04-24 13:57:01,862 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: d@@@@@000   0-F1\r
2023-04-24 13:57:01,912 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: None
2023-04-24 13:57:01,914 - adapters.fakesmartrow.fakesmartrowble - WARNING - no data from SmartRow interface
-->e@@@@E000000 w61
2023-04-24 13:57:01,964 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: e@@@@@000000 w5C\r
-->f@@@@E---00!--0D
2023-04-24 13:57:02,014 - adapters.fakesmartrow.fakesmartrowble - INFO - Sending: f@@@@@---00!--08\r
```

dan06, 2023-04-24:
@gabrykkk is your phone Android or IOS?  
Do you have the latest version of the SmartRow app?
Does the original setup from inonoob work?

Below is what I get in the piroflow log when everything works.  Can you post a similar complete log from startup?

2023-02-02 15:26:20,799 - __main__ - INFO - S4 not selected
2023-02-02 15:26:20,805 - __main__ - INFO - interface smartrow will be used for data input
2023-02-02 15:26:20,813 - __main__ - INFO - Smartrow Interface started
2023-02-02 15:26:20,815 - __main__ - INFO - SmartRow passthrough is enabled
2023-02-02 15:26:20,844 - __main__ - INFO - Start SmartRow Passthrough BLE Advertise and BLE GATT Server
2023-02-02 15:26:20,854 - __main__ - INFO - Bluetooth service not used
2023-02-02 15:26:20,867 - adapters.fakesmartrow.fakesmartrowble - INFO - Waiting for real SmartRow to connect
2023-02-02 15:26:20,880 - __main__ - INFO - Start Ant and start broadcast data
2023-02-02 15:26:20,950 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2023-02-02 15:26:21,171 - adapters.smartrow.smartrowreader - INFO - starting discovery
2023-02-02 15:26:44,374 - root - INFO - found SmartRow
2023-02-02 15:26:44,380 - root - INFO - xx:xx:xx...
2023-02-02 15:26:44,393 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2023-02-02 15:26:44,475 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2023-02-02 15:26:54,960 - adapters.smartrow.smartrowreader - INFO - Connected to [xx:xx:xx...]
2023-02-02 15:26:56,393 - adapters.smartrow.smartrowreader - INFO - Resolved services [xx:xx:xx...]
2023-02-02 15:26:56,398 - adapters.smartrow.smartrowreader - INFO - 	[xx:xx:xx...] Service [00001234-0000-1000-8000-00805f9b34fb]
2023-02-02 15:26:56,404 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001236-0000-1000-8000-00805f9b34fb]
2023-02-02 15:26:56,410 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001235-0000-1000-8000-00805f9b34fb]
2023-02-02 15:26:56,415 - adapters.smartrow.smartrowreader - INFO - 	[xx:xx:xx...] Service [00001801-0000-1000-8000-00805f9b34fb]
2023-02-02 15:26:56,479 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
2023-02-02 15:26:57,619 - adapters.smartrow.smartrowtobleant - INFO - Received SmartRow challenge request!
2023-02-02 15:26:57,627 - adapters.smartrow.smartrowtobleant - INFO - Sending SmartRow challenge response!
2023-02-02 15:27:00,535 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2023-02-02 15:27:00,801 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using /org/bluez/hci0
2023-02-02 15:27:00,904 - adapters.fakesmartrow.fakesmartrowble - INFO - Registering GATT application...
2023-02-02 15:27:00,955 - adapters.fakesmartrow.ble - INFO - GetManagedObjects
2023-02-02 15:27:00,970 - adapters.fakesmartrow.ble - INFO - GetAll
2023-02-02 15:27:00,993 - adapters.fakesmartrow.ble - INFO - returning props
2023-02-02 15:27:01,007 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2023-02-02 15:27:01,020 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered
2023-02-02 15:27:06,614 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification
2023-02-02 15:27:06,644 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received $
2023-02-02 15:27:06,653 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=1
2023-02-02 15:27:06,734 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received #
2023-02-02 15:27:07,814 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received $
2023-02-02 15:27:07,819 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=2: Sending challenge
2023-02-02 15:27:07,826 - adapters.fakesmartrow.fakesmartrowble - INFO - Generated KEYLOCK=BD120199
2023-02-02 15:27:07,904 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received 
2023-02-02 15:27:07,910 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=3: Receive count=1
2023-02-02 15:27:09,839 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received $
2023-02-02 15:27:09,845 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=3: Receive count=2
2023-02-02 15:27:09,976 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received e
2023-02-02 15:27:09,987 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=3: Receive count=3
2023-02-02 15:27:10,065 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received c
2023-02-02 15:27:10,088 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=3: Receive count=4
2023-02-02 15:27:10,200 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received 7
2023-02-02 15:27:10,205 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=3: Receive count=5
2023-02-02 15:27:10,290 - adapters.fakesmartrow.fakesmartrowble - INFO - App not connected and received c
2023-02-02 15:27:10,296 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=3: Receive count=6
2023-02-02 15:27:10,316 - adapters.fakesmartrow.fakesmartrowble - INFO - Connect state=4: Connected

gabrykkk, 2023-04-25:
Hi, sure:
is your phone Android or IOS? iOS with latest version
Does the original setup from inonoob work? yes

Here is the log after completely clean installation, attached with "debug" also here:
[piroflowdebug.txt](https://github.com/inonoob/pirowflo/files/11324240/piroflowdebug.txt)
I couldn't find where I get the error "invalid literal for int() with base 10: ''" yet!
Thanks!

```
root@raspberrypi:~/pirowflo/src# python3 waterrowerthreads.py -i sr -a
2023-04-25 09:49:25,754 - __main__ - INFO - S4 not selected
2023-04-25 09:49:25,756 - __main__ - INFO - interface smartrow will be used for data input
2023-04-25 09:49:25,757 - __main__ - INFO - Smartrow Interface started
2023-04-25 09:49:25,758 - __main__ - INFO - SmartRow passthrough is enabled
2023-04-25 09:49:25,764 - __main__ - INFO - Start SmartRow Passthrough BLE Advertise and BLE GATT Server
2023-04-25 09:49:25,766 - __main__ - INFO - Bluetooth service not used
2023-04-25 09:49:25,767 - adapters.fakesmartrow.fakesmartrowble - INFO - Waiting for real SmartRow to connect
2023-04-25 09:49:25,768 - __main__ - INFO - Start Ant and start broadcast data
start search for dongle
2023-04-25 09:49:25,779 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2023-04-25 09:49:25,794 - adapters.smartrow.smartrowreader - INFO - starting discovery
Using Dynastream Innovations dongle
calibration
set Networkkey:b'\x00\xxx xxxE'
create Channel
2023-04-25 09:49:30,985 - root - INFO - found SmartRow
2023-04-25 09:49:30,986 - root - INFO - xx:xx:xx
2023-04-25 09:49:30,989 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2023-04-25 09:49:31,024 - adapters.smartrow.smartrowreader - INFO - Preferred SmartRow adapter is hci1
2023-04-25 09:49:37,205 - adapters.smartrow.smartrowreader - INFO - Connected to [xx:xx:xx]
2023-04-25 09:49:38,263 - adapters.smartrow.smartrowreader - INFO - Resolved services [xx:xx:xx]
2023-04-25 09:49:38,264 - adapters.smartrow.smartrowreader - INFO - 	[xx:xx:xx] Service [00001234-0000-1000-8000-00805f9b34fb]
2023-04-25 09:49:38,264 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001236-0000-1000-8000-00805f9b34fb]
2023-04-25 09:49:38,265 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001235-0000-1000-8000-00805f9b34fb]
2023-04-25 09:49:38,265 - adapters.smartrow.smartrowreader - INFO - 	[xx:xx:xx] Service [00001801-0000-1000-8000-00805f9b34fb]
2023-04-25 09:49:38,299 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
starting heart beat
2023-04-25 09:49:42,315 - adapters.fakesmartrow.fakesmartrowble - INFO - Real SmartRow connected
2023-04-25 09:49:42,369 - adapters.fakesmartrow.fakesmartrowble - INFO - Fake SmartRow using /org/bluez/hci0
2023-04-25 09:49:42,383 - adapters.smartrow.smartrowtobleant - INFO - Received SmartRow challenge request!
Checksum GOOD
[13, 102, 57, 56, 49, 13]
2023-04-25 09:49:42,386 - adapters.smartrow.smartrowtobleant - INFO - Sending SmartRow challenge response!
2023-04-25 09:49:42,391 - adapters.fakesmartrow.fakesmartrowble - INFO - Registering GATT application...
2023-04-25 09:49:43,004 - adapters.fakesmartrow.ble - INFO - GetManagedObjects
invalid literal for int() with base 10: ''
f
2023-04-25 09:49:43,017 - adapters.fakesmartrow.ble - INFO - GetAll
2023-04-25 09:49:43,018 - adapters.fakesmartrow.ble - INFO - returning props
2023-04-25 09:49:43,020 - adapters.fakesmartrow.fakesmartrowble - INFO - GATT application registered
2023-04-25 09:49:43,030 - adapters.fakesmartrow.fakesmartrowble - INFO - Advertisement registered
2023-04-25 09:49:58,845 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification
2023-04-25 09:50:56,803 - adapters.fakesmartrow.fakesmartrowble - INFO - Ending notification
2023-04-25 09:50:56,804 - adapters.fakesmartrow.fakesmartrowble - INFO - Resetting app connection
could not convert string to float: '0006\x04'
c000270000006>'
2023-04-25 10:02:04,887 - adapters.fakesmartrow.fakesmartrowble - INFO - Starting notification
2023-04-25 10:02:23,966 - adapters.fakesmartrow.fakesmartrowble - INFO - Ending notification
2023-04-25 10:02:23,968 - adapters.fakesmartrow.fakesmartrowble - INFO - Resetting app connection
^C2023-04-25 10:06:02,337 - __main__ - INFO - Quit gracefully program has been interrupt externally - exiting
```

dan06, 2023-04-26:
@gabrykkk do you have an Android device to test with?  The log you provided looks ok.  I don't have any IOS devices.

FYI, the invalid literal is not an issue. It happens during startup when the code gets partial data from the SmartRow.

joedinkle, 2023-10-08:
@dan06 thanks for your work on this! I'm using your fork. Everything works, but I'm curious if you figured a way to run this with the Bluetooth dongle plugged in at boot. It's the only issue I have. It's minor, but I'd like to get this fixed. Thanks!

dan06, 2023-10-08:
@joedinkle my system works fine with the current code (of course :) ).  I have a Pi Zero W2 with a Bluetooth dongle plugged in all the time.  I normally power up my system before use, then start the PiRowFlow program using the OLED screen. I then shut down the Pi when I'm done.  This works 99% of the time.  Very occasionally, the screen won't start and I have to repower the Pi.

Can you post the log for when it works and when it doesn't work?  I may be able to suggest how you can fix it for your system.  What hardware do you use?  Does the OS see the BT dongle if you plug it in before boot?

stampede79, 2023-10-23:
> @stampede79 I don't think the SR App is completing the connection to the Pi. I had some issues on my system.
>
> Check the StartNotify() function in fakesmartrowble.py. It contains a couple of debug [...]

Hi dan06 again :-)

unfortunately I ran into the same issue after reinstalling Pirowflo. As you said I added a println for debugging the code and it seems that the StartNotify() Function is never properly started. Can you still remember how to fix this ?

Thx in advance
Jan

dan06, 2023-10-24:
> 
> unfortunately I ran into the same issue after reinstalling Pirowflo. As you said I added a println for debugging the code and it seems that the StartNotify() Function is never properly started. Can you still remember how to fix this ?
> 
> Thx in advance Jan

@stampede79 , This is what you posted above as having fixed the issue:
I did a complete reinstallation of rasberry os and pirowflo.
After restarting the service i saw, that the wrong adapters were used for the connections (SR -> Piroflow internal Bluetooth and Pirowflo -> SR-APP dongle). So I changed the def get_sr_preferred_adapter() method in src/adapters/fakesmartrow/ble.py and src/adapters/smartrow/smartrowreader.py (USB -> UART and the other way round). First test seems to run...

Have you tried that again?

stampede79, 2023-10-25:
> @stampede79 , This is what you posted above as having fixed the issue: I did a complete reinstallation of rasberry os and pirowflo. After restarting the service i saw, that the wrong adapters were [...]

Hi dan06,
Thx for your fast answer.
I reinstalled freshly and changed USB-> UART in the files to make oirowflo available. Now I see the pirowflo adapter at my watch but not in the SR. I have added the debug-line in the StartNotify()-funktion (add a println), which did not occur in the output. So this function seems to be suspicious concerning my issue ?!
Regards

[9 more replies]