inonoob/pirowflo #32: ATN+ dongle choise

zpukr, 2021-02-24:
You recommend avoid Cyclone when choosing Ant+ dongle. I could not find a single adapter on the Cyclone chip (is this exist?) in google however. Could you tell more correct which specific models should be avoided and which ones are recommended for use? Thanks!

inonoob, 2021-02-24:
Hey @zpukr  

As I got the code for Ant+ from the project FortiusANT, this is where I got my information about the issue with Cyclone sticks. 

check this link. [link](https://github.com/WouterJD/FortiusANT/issues/61) 

I bought the following [link](https://www.amazon.de/-/en/Transmitter-Receiver-Transfer-Compatible-TrainerRoad/dp/B08DCN99Z7/ref=sr_1_2?dchild=1&keywords=chileaf+ant%2B&qid=1614159676&sr=8-2) and didn't have any issues 

I hope it helps you. 

Could you please tell me your setup, are you using SmartRow or S4 ?

zpukr, 2021-02-24:
Things are a little more complicated. I have a сhinese clone S4 and am trying to equip it with bluetooth and ant + adapter, connection with my watch garmin 5 plus is of interest first of all (hope you will implement ErgIQ). But at the moment I am at the early stage of purchasing components for project. 
Thanks for the answer!

inonoob, 2021-02-24:
Hey @zpukr, 

So from what, I got as Information, the Fenix 5 is not working as the native rowing app don't get any update. And in order to implement ErgIQ, I would need to get in contact with them. I haven't found it yet. As of today, I would say garmin fenix 5 is not supported yet. #20 

As for your Chinese Rower. Might you give some more feedback. 
I could only see two solution to integrate it: 
- You either get a SmartRow and fit it into the Chinese Rower (easier but more expensive) 
- You know how the Chinese Monitor is outputting data and it would be possible to add it as input. (cost less but must be implemented ) 

maybe upload some picture of your rower. And if you could find data, that would be great to have a look.

zpukr, 2021-02-24:
1. I read the links you provided and see a recommendation to avoid CYCPLUS adapters but not Cyclone, which is probably why there was a misunderstanding.
2. Do not be distracted by my problem, please. A chinese monitor is complete junk, and I cannot extract any useful information from it. Rower like this https://www.amazon.com/Incline-Fit-Rowing-Machine-Monitor/dp/B07FQT3WXS/ref=cm_cr_arp_d_bdcrb_top?ie=UTF8
But for me it is not too much of a problem to create my own adapter on Arduino that will read data from the speed sensor and transmit it to a UART or bluetooth. At the first stage, I plan to transfer data via bluetooth in order to use your project without modifications. But repeat that I am interested in the integration with the garmin Fenix 5 Plus, there are no plans to buy the Fenix 6 yet, it is too expensive :)

inonoob, 2021-02-24:
Oh !!! I haven't see this. Many thanks for pointing out my error !! 

In order to connection from your Bluetooth module to PiRowFlo via Bluetooth. PiRowFlo then needs to act as client with the modified profile . Today we use the SmartRow bluetooth profile. But what you can do is to create an Comm module with the Bluetooth Fitness equipment profile and then PiRowFlo needs an adapted GATT client to talk to your device. Or PiRowFlo needs to understand the UART data. 

have you checked the two projects here on github using the Sensor ? 
Use ESP32 but uses the USB port.  [link](https://github.com/vibr77/wr_s4bl3_samd)
Use Raspberry pi GPIO [link](https://github.com/andresol/waterrower-a1-logger)

zpukr, 2021-02-24:
Yes, I plan to mix two existing projects WaterRino (reading data from the sensor) https://github.com/adruino-io/waterrino and wr_s4bl3_samd (transmitting via bluetooth, disguising as a regular S4) https://github.com/vibr77/wr_s4bl3_samd. I think this is a simple task :)

inonoob, 2021-02-24:
That 's cool have you an idea about your timeline ? Please keep me in the loop.

zpukr, 2021-02-24:
I'm not sure about the timing of these plans, maybe a couple of months. But your implementation of ErgIQ support can speed up this process :) As soon as I have some interesting results, will definitely inform you about it!

inonoob, 2021-02-24:
Can you find out who it the autor of ErgIQ :)

zpukr, 2021-02-24:
I am not familiar with the author, but you can contact him on this page: 
https://apps.garmin.com/en-US/apps/2fc7c56b-f7f8-4fc4-b92e-8d3bde99f693
or here https://forums.garmin.com/developer/connect-iq/f/showcase/4281/erg-iq

inonoob, 2021-02-24:
Thanks, 

I just sent them an e-mail. Let's see what they say :)

inonoob, 2021-02-24:
I close the issue for now

closed 2021-02-24 by inonoob

zpukr, 2021-02-24:
You can also contact the author of another interesting application with functionality similar to ErgIQ. For this app the datafield version is available also!
https://apps.garmin.com/en-US/apps/74afbe31-c6c4-47ee-8776-3b9778cb676b#0

inonoob, 2021-02-24:
Have you tried it ?

zpukr, 2021-02-24:
Of course not! I don't have a concept2 rowing machine, so I installed the app, but there is no way to test it. I think that by implementing support for either "ErgIQ" or "AK Concept2 Rowing", we will get support for both (and much more!), since they use the same concept2 ANT+ protocol. I suggest contacting this author as an alternative in case the author of ErgIQ does not find the time or opportunity to help you

zpukr, 2021-02-28:
I've already done the Arduino part https://github.com/zpukr/ArduRower. But further development will stop, since firstly I do not know what a Raspberry Pi and python is, and secondly, I do not have a Raspberry Pi (already purchased, but bought in China, which means a long delivery). I realized that bluetooth connection with PiRowFlo is not an option for me, since I emulate an S4 comm module, not SmartRow. Therefore, I have to connect to PiRowFlo via UART, I hope that I directly connection (not USB) with TX/RX pins not a problem. Who should initiate the dialog, Arduino or PiRowFlo? I'll figure it out when I get my hands on the Raspberry Pi

inonoob, 2021-02-28:
I see two solution. To either implement the possibility to connect to your device via Bluetooth or use the serial ports. But either ways a listener in pirowflo must be programmed.

zpukr, 2021-02-28:
We get the benefits of using serial port - the Raspberry Pi's internal bluetooth adapter can be used to connect to various applications, there is no need to purchase an additional dongle. But I will repeat once again PiRowFlo interests me exclusively as an ANT+ transmitter (for use with ErgIQ or similar applications for garmin). ANT+ for Arduino is possible, but this is a difficult task (at least for me), bluetooth, on the contrary, is quite simple and chargeless on ESP32, I do not need PiRowFlo for it.

zpukr, 2021-03-07:
After a week of brainstorming, my problem was solved without using a Raspberry Pi! I had to write my own ConnectIQ DataField namely ArduRower-IQ. You can see screenshots on my page https://github.com/zpukr/ArduRower
DataField is not publicly available yet, this is my first application for ConnectIQ so additional testing is needed. But if you are interested, I can send it for you for testing, DataField should also work with PiRowFlo (not only ArduRower) when connected to bluetooth using the S4 Monitor protocol. Many Garmin watches are supported, starting with Fenix 5 and newer, Forerunner 245 and newer, etc

inonoob, 2021-03-07:
Hey, that's very good news. And really like your design. I think a microcontroller is for things like this superior as the Raspberry pi. The thing with the Raspberry pi is , I know how to code python and I know Linux. I have done some things with Arduinos and esp8266 but my c skills are not the best. :).

zpukr, 2021-03-07:
Must work, minimal ConnectIQ SDK version is 3.1.x and Vivoactive 3 in this list

zpukr, 2021-03-09:
I have not received an any response from you if you are interested in using my datafield with PiRowFlo. I posted a ready-to-use datafield in the https://github.com/zpukr/ArduRower/tree/main/firmware folder, I'm not sure if it can be used with vivoactive 3 (but vivoactive 3 Misic seems ok), so this is the version for Fenix 6

zpukr, 2021-03-09:
Yes. Download .PRG-file from my folder and put it in that GARMIN/APPS folder via USB, add datafield to any activity. An sideloaded application (not from the store) does not have the right to reciord to the .fit-file, but you should see the distance and speed on the screen of watch

zpukr, 2021-03-09:
By the way. Support for the original S4 with a comm module can also be added, but I need the first two bytes of the packet that it sends via bluetooth. Real bit-to-bit decoding is too hard for me, so hardcoded only. But I do not understand why I should do this and not a waterrover company with billions of budgets, support for open source projects as yours is a completely different matter

inonoob, 2021-03-09:
It such a shame that Waterrower has no luck with there IT products. The Com module is not very reliable. It now even sells in Germany for 69€ !!! If you check what concept 2 is doing with Ant+ support, bluetooth and the logbook. They are trying. The S4 monitor for 250€ is also not acceptable as a price. That thing is from 2009 and just a microcontroller with a big screen. 
Let's hope that the SmartRow kicks Waterower out of the stone age.

zpukr, 2021-03-11:
https://apps.garmin.com/en-US/apps/0e10bae1-9753-4915-b856-040d0cbdd82a