inonoob/pirowflo #33: Shutdown of RasPi via Supervisor

twollweber, 2021-02-24:
While starting the Raspberry can be done by simply powering it and as PiRowFlo comes up mostly by itself, there is no simple means to shut the Raspberry down. It currently involves an SSH connection or local access. This is no ideal solution for non-geek rowers (i.e. family members).

The graceful shutdown of the Raspberry could be initiated from the web page of the Supervisor.
As the rower will use a Wi-Fi enabled device anyways (to track the data) and he/she is able to connect and start the proper script via the web page, a graceful shutdown from the same page would be an elegant way to end the training.

Once the OLED + keys HAT will be fully integrated, the "graceful shutdown" function will certainly be available through that means. But even then, the least expensive setup for a number of users (myself included) is the bare Raspberry and those rowers can live with a click on a web page.

**Additional context**
This could even be extended to add MQTT client capabilities in order to integrate the device in home automation systems and to allow remote control of PiRowFlo through other means.

inonoob, 2021-02-24:
lessons learned: 

if you forget in the supervisor conf the flag autostart=false then you are in for a shutdown loop :)

inonoob, 2021-02-24:
Implemented !

closed 2021-02-24 by inonoob