inonoob/pirowflo #34: Issue with some apps not connecting to PiRowFo

inonoob, 2021-03-12:
Some apps don't connect to PiRowFlo via bluetooth. 

if you try the app Waterrower connect, the app will see PiRowFlo but won't connect to it 

the problem is that PiRowFlo or better say bluez is using Bluetooth classic at the same time as bluetooth Low energy. That is what makes all the apps going crazy and for some won't connect because they expect to only see a bluetooth low energy device. 

In order to change this, we must start the bluetooth dongle in `LE-only` mode. But here is the catch if you use more than one bluetooth dongle, this is the case when you use smartrow, the setting only changes the setting for one dongle. The other still have the issue with bluetooth classic and bluetooth low energy enable. 

How to correct this: 

1. for S4 Monitor only,  only the `/etc/bluetooth/main.conf` has to be updated. It must be switch from  `dual` to `le`

```
# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
#ControllerMode = dual
```

2 for SmartRow the situation is more difficult. In order to correct that we need to download the latest version from bluez which is 5.56. Then we need to edit the main.conf: 

```
static int get_mode(const char *str)
{
	if (strcmp(str, "dual") == 0)
		return BT_MODE_DUAL;
	else if (strcmp(str, "bredr") == 0)
		return BT_MODE_BREDR;
	else if (strcmp(str, "le") == 0)
		return BT_MODE_LE;

	error("Unknown controller mode \"%s\"", str);

	return BT_MODE_DUAL;
}
``` 

we change the default value from being `BT_MODE_DUAL` to only be `BT_MODE_LE` 

then we have to recompile bluez 5.56 and install it on pirowflo to replace bluez 5.50. 

This must be of course automated via the `install.sh` script which will lead to a first installation time being 15 to 20 min

mittelmo, 2021-03-14:
Hey inonoob. 

Great work on the project. 
I've been trying to set up the system with the WaterRowerConnect App. 
The Waterrower (s4 monitor) is recognized by the app and get's connected to it. However, no data seems to be arriving at the app. 
Your code also gives a warning: 
adapters.ble.waterrowerble - WARNING - no data from the s4 interface 

Any idea why this is happening? 

Thanks, 
mittelmo

inonoob, 2021-03-14:
Hey 

2 Things have you tried with coxswain if it is working? And when you connect the S4 and PiRowFlo start do you hear the beep from the S4 which is been reseted by PiRowFlo? 

I will check this evening if I have the same issue

inonoob, 2021-03-14:
Hey, 

So I checked with Waterrower Connect. And it doesn't get the data. On iOS it is working but android it doesn't. I have check with:
-  coxswain ==> I see all the values and It's working
- nRF ==> I see all the values and It's working 

Waterrower Connect is not working and I don't know why. I can see that it is asking for data and that the data are there from the S4 but I don't know why the Android version isn't working !

inonoob, 2021-03-15:
Hey all,

The following progress has been made:
I created an install script in order to download the latest bluez version and have all the configuration to compile and then install it.
I also work on integrating the bluez installation script in the main install.ah script. This will check if the bluez version is lower as 5.56, if yes then get, compile and install bluez.

venturanc, 2021-03-18:
Hi, hopefully I'm not looking in the wrong place, but I don't see that the project with the new install script is available here on github yet, right?

Without the new install script and without having updates for bluez or config changes, I'm having the following experience with troubleshooting (I think this is expected based on your comments above):

Setup: Raspberry Pi 3B, latest Raspberry Pi OS installed just last week, with a Garmin Ant+ (Dynastream) ant+ adaptor, connected to Waterrower S4 monitor.  Started PiRowFlo in BLE/Ant+ mode. Verified that all BLE-connecting apps are disconnected and force-quit when attempting to trial a different app, and restarted multiple times to make sure that one connected app isn't stopping another app from making the connection.

- Ant+ trainer connects to Garmin Fenix 6 watch and passes all data points into the built-in indoor rowing app.
 -nRF on android can connect and see
- Zwift on iPad/iOS sees PiRowFlo as a trainer and power meter source and seemingly connects but never sees more than zero watts, when you start rowing (at the same time that the Fenix 6 watch is receiving the data successfully via Ant+). 
- Waterrower Connect on Android sees PiRowFlo but can't connect. 

Haven't tried Coxswain on android, or Waterrower Connect on iOS yet. 

Thanks again!  Seriously, even just getting the Ant+ data is a game-changer!

inonoob, 2021-03-18:
Hey, 

So the current status is: 
- the installation script for new bluez 5.56 is ready but I still fight to get all the dongles in Bluetooth low energy mode. I can't seems to make it work after a reboot. It always goes back to Bluetooth classic and Bluetooth low energy. I already made a post on the bluez GitHub page. 
- that zwift is not working could have different reasons. Zwift sees PiRowFlo as fitness equipment but expect to have the indoor bike profile and not a rowing profile. That might be a possibility. Or it's the same issue which I have with waterrower connect. Which is I get a connection but no data.
- With the new config for bluez I can connect to waterrower connect on Android but when I row waterrower connect don't see any data. But if I check with nRF app, I see the data. So I currently concluded something is still missing for waterrower connect on Android. The strang thing is on iOS it is working. 

I'm still fighting with Bluetooth on Linux. I never had though to start debugging Linux kernel stuff, modify source code and compile things my self just to make waterrower smart :) 

I haven't upload or commit in a long time as this topics eats a lot of time !

inonoob, 2021-03-22:
Hey, 

so this is now I think third week in a row which I try to setup Bluez starting all the Bluetooth dongle in Bluetooth low energy only. 

Bluetooth for Raspberry pi is a rabbit hole !! 

So the vanilla Bluez implementation is a different one from the Raspberry pi one. The raspberry pi version uses old deprecated tools like `hciattach` and also the source code of `hciattach` has been changed.

Vanilla Bluez (not modification to the source code) 
- Bluez 

Raspberry pi Bluez: 
- Bluez (with modified deprecated source code tool `hciattach`) has been replace in bluez with `btattach`
- pi-bluetooth 
- Bluez-firmware 

I tried to set the raspberry pi version of Bluez5.50 to only start in Bluetooth LE only mode. I could get it work! Like I said I might have look on the wrong spot but after 3 weeks I came to the conclusion I want to control the boot process for bluetooth module. 

Here is the close to finish solution: 

- make a copy of `sudo cp /user/bin/hciattach /user/bin/hciattachrpi`
- purge the raspberry pi `bluez5.50` version which will also remove `pi-bluetooth`  with `sudo apt-get purge bluez`
- start my created install script `sudo ./install-bluez.sh` 
      - download the latest version of `Bluez` with the latest commit 
      - download needed dependency to build `Bluez` 
      - configure before making the source `./configure` 
      - make it (needs 15min to 20minto build) 
      - and then install `Bluez` on the raspberry pi  
      - set `ControllerMode = le` to only start in Bluetooth low energy mode
      - set `--experimental` to `exec`part of the  `/etc/systemd/system/bluetooth.service` 
      - start Bluetooth with systemd `sudo systemctl enable bluetooth.service`
      - download `btuart` from pi-bluetooth repo 
      - install `btuart` into `/user/bin/btuart`
      - create new service to start internal raspberry pi Bluetooth `btuart`
      - reboot 
- Then all should be installed and the raspberry pi has all the Bluetooth dongle in low energy configuration. 

So further testing is needed. 

best regards

inonoob, 2021-04-11:
Perfect, nice to hear that the project is lifting of :)

jaquoh, 2021-12-02:
Very nice project! I just tried it out, with a pi zero w and everything works fine. Waterrower beeps and I can receive the data via nrf connect on android. Unfortunately kinomap (which I like to use) does recognize the PiRowFlo but does not successfully connect to it. Can you share the `install-bluez.sh` script you mentioned? I would be happy to test if this fixes it. :)

JBlaak, 2022-01-21:
Hi! Loving this project! I'm actually one of the developers behind the WaterRower Connect app, could you contact me at connect@waterrower.com so we can look into getting this project to work with both our Android and iOS app.

DANgerous25, 2022-03-14:
Hi, I'm wondering if the issues I'm experiencing relate to what you describe in this thread. I have tried with two apps:

1) Kinomap on IOS - in Kinomap using the FTMS protocol. I had this working once, which is when I reported it as good after an initial test. However, I can't get it working again. I can get Kinomap to see pirowflo until Smartrow connects. But either way, even though I can see data feeding into pirowflo, Kinomap doesn't get the data. Just zeros for everything.

2) EXR - EXR this week released version 1.6 on IOS which includes FTMS compatibility. When trying to connect to Pirowflo, EXR sees Pirowflo as a bluetooth device and allows you to pair. However, when you agree to pair you are prompted again to pair several times before it stops asking and appears to be paired. It's almost like the connection drops quickly. Even after saying "yes" to pair several times, EXR then loses connection and you have to initiate the pairing again. This all happens very quickly, I have not been able to get EXR to see any SmartRow data.

Admittedly EXR is very new, so might not work anyway, but I just wonder if the symptoms are the same for Kinomap and potentially other apps.

Also on the topic of EXR, I contacted them to see if there would be connectivity with pirowflo and smartrow. They said that they once had contact with the pirowflo developer but didnt' hear back. JFYI. 

`pi@pi1:~ $ bluetoothd -v
5.55`

Do I need to upgrade bluetooth, and if so how do I do it? There are a couple of posts above but not sure which is current and how lasting the effects would be.

bchltz, 2022-12-30:
After some months of not using the Pi I can't connect to Waterrower Connect any longer. Potentially due to app updates (I´m running 2.12.5 on iOS. I was able to connect to Kinomap, so the Bluetooth connection seems to work.

jcbigears, 2023-04-24:
Thanks for answering anyway. I have a spare Pi lying around so thought I'd give it a go. I'll let you know.
John

DANgerous25, 2023-04-24:
If I recall correctly, the BlueTooth dongle that you use makes a big difference. I can't remember which one I got in the end, I think it was recommended in the project specs.

Good luck!