inonoob/pirowflo #35: [BUG] Cannot Connect ANT chest strap

zwiwwelsupp, 2021-03-28:
Hi,
not sure whether it is a real software "bug" or a misunderstanding. There is a picture that it is possible to connect ant device such as a chest strap. When starting there is no note about the connection of external ANT chest strap. What do I need to do for connecting the chest strap?

Regards

Johannes

inonoob, 2021-04-11:
Hey sorry for the late post, 

So this is not possible yet, as i haven't merge the code in order to do so. That is on my to do list. I need to release the bluetooth bluez part first.

nickbits1024, 2021-12-14:
I'm not sure how @inonoob feels about 3rd party contributors but I have been working on an update to add this.  I have it working but am testing it a bit more.  It also enables the BLE HRM service.
The other feature I'm trying to add is to make pirowflo also advertise itself as an indoor bike so it works with vzfit.  My HRM code is a bit jumbled with that update as well so either I need to finish it or separate it out.

inonoob, 2022-03-05:
Hey @nickbits1024,

Have you adapted the code ? 

BR

nickbits1024, 2022-03-05:
> Hey @nickbits1024,
> 
> Have you adapted the code ?
> 
> BR
yes. i have the hrm update isolated in its own commit . i can push it to a branch if you want to give me contributor access then you can do what you want with it. 
i couldn’t get the bluez stack to work with the bike  sensor profile (in the app i wanted to use it with) so i ended up making a whole another project based on esp32 and haven’t been using pirowflo.  i am going to post that online as well one i get some docs written up.