inonoob/pirowflo #37: Auto Start feature

kaz911, 2021-05-01:
I would love to see an "auto-start" feature please 

I know I could start the waterrowerthreads.py from a script but that start does not "pass status through" to the web interface - so can't be shut down from web interface.

So in the WEB UI - make a selection of which "option" version you want to autostart - and start it upon boot. From the Web UI - be able to shut down the version and/or change/start other "option versions"

Benefits: You do not need a screen - or device with web interface to start the service and it is ready when ever you turn on your Pi.

twollweber, 2021-05-02:
I edited the _supervisord.conf_ and put _autostart = true_ for the service which I use. Now all that's left to do is connect the RasPi via USB to my S4, power it up and wait for the second beep from the S4.
After the training is finished I open the WebUI via _http://pirowflo.local:9001_ and shut down the RasPi.

kaz911, 2021-05-03:
that helps :) but it would still be nice to add a few radio buttons to the status web page where you can select "autostart" 

I bought a display but the crappy display shorts out the PI when screwed in (posts delivered are not tall enough for my PI3 - so whole display PCB-- warps when the screws are in it - and when the display warps it shorts out...)

So if you have PI3 - do not buy this display branded Waveshare: https://www.amazon.co.uk/gp/product/B078D6NXFM/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

inonoob, 2021-05-19:
Hey, 

The thing with the webpage, I use here a framework which I can't really currently change. 

Best Regards