inonoob/pirowflo #39: [BUG] Debugging ANT+ connectivity

temancl, 2021-05-19:
I'm having trouble pairing my Garmin 6x Pro - no new adapter shows up. I've tried a couple different adapters, including one I am using a separate (and similar) spin bike converter.

I've verified with the BT LE config fix that WaterRower Connect works OK, so the Pi is connected to the S4, transmitting data, etc. 

Any suggestions on diagnosing missing ANT+ systems?

[    4.910129] usb 1-1.4: New USB device found, idVendor=0fcf, idProduct=1008, bcdDevice= 1.00
[    4.916059] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.919084] usb 1-1.4: Product: ANT USBStick2
[    4.922043] usb 1-1.4: Manufacturer: Dynastream Innovations
[    4.925145] usb 1-1.4: SerialNumber: 123

temancl, 2021-05-19:
It's permissions, I should have just tried running the script with sudo. I'll figure out the missing piece and reply.

inonoob, 2021-05-19:
Hey 

So if you start the PiRowFlo script with the option -b -a you can see PiRowFlo on your watch. 

We should ensure that the install.sh has place the relevant data into the folder:

/etc/udev/rules.d/99-garmin.rules

Do you use the default user pi ? 
Do you use a raspberry pi? 

Best rergards

temancl, 2021-05-19:
Thanks, I think I just figured it all out. I'll re-test in the morning with the rower.

1. I didn't run the install script with sudo, so it didn't install the udev rules. These rules also depend on the ANT transmitter being connected at install time. Consider testing for sudo & warning about having all your USB stuff connected? And/Or having a looser generic udev rule that's installed unconditionally?
2. The logging didn't produce any obvious error messages that it didn't find an ANT transmitter. Also consider erroring out if you specify -a and there's no transmitter attached so it shows up in the logs?

inonoob, 2021-05-19:
Hey 

1. It's a clever idea to check for root!  I haven thought about that one. The thing is some people don't have the ANT+ dongle at the beginning or use it at all. In addition, I would like to do some more generic udev rules but here is the point, I have to create the rule depending on the correct productID 1008,1009 or 1004 which depends on the USB ant+ dongle you use. 

2. The point checking for the dongle before the start of the script  is also a clever idea. I need to check how I can ensure that the dongle is connected and installed properly. 

best regards

temancl, 2021-05-19:
Could you just install the udev rules for all known ANT+ pids? If it's just permissions it shouldn't matter if you have a superset of rules.

temancl, 2021-05-20:
Tested out great! This is resolved, everything here is just suggestions now. Thank you!

closed 2021-05-20 by temancl