inonoob/pirowflo #41: SmartRow, no connection

brokenpi, 2021-08-19:
Thanks for your great work! I used pirowflow together with the S4 Monitor, but now i upgraded for smartrow.
So I removed the S4 Monitor at all.
Pirowflow is running on a fresh install of Pi OS lite on Pi Zero W. After installing the smartrow, i plugged a blutetooth stick (Hama USB Bluetooth Adapter 4.0, 00053188) to the Pi and rerunned the install.sh. Then i startet pirwoflow SR bluetooth only.
Sadly no connection is established, the smartrow keeps blinking.

The log looks like this: 
`2021-08-19 15:57:59,817 - __main__ - INFO - Smartrow Interface started
2021-08-19 15:57:59,828 - __main__ - INFO - Start BLE Advertise and BLE GATT Server
2021-08-19 15:57:59,836 - __main__ - INFO - Ant service not used
2021-08-19 15:58:00,014 - adapters.smartrow.smartrowreader - INFO - starting discovery
2021-08-19 15:58:00,075 - adapters.ble.waterrowerble - INFO - Registering GATT application...
2021-08-19 15:58:00,851 - adapters.ble.ble - INFO - GetManagedObjects
2021-08-19 15:58:00,982 - adapters.ble.ble - INFO - GetAll
2021-08-19 15:58:00,988 - adapters.ble.ble - INFO - returning props
2021-08-19 15:58:01,000 - adapters.ble.waterrowerble - INFO - GATT application registered
2021-08-19 15:58:01,011 - adapters.ble.waterrowerble - INFO - Advertisement registered
Update Waterrower Data
2021-08-19 15:58:16,520 - adapters.ble.ble - INFO - Default StartNotify called, returning error
2021-08-19 15:58:16,601 - adapters.ble.waterrowerble - WARNING - no data from s4 interface` 

Sometimes it showed a discovery:
`2021-08-19 15:50:55,547 - adapters.ble.waterrowerble - INFO - Registering GATT application...
2021-08-19 15:50:55,712 - adapters.smartrow.smartrowreader - INFO - starting discovery
2021-08-19 15:50:55,780 - adapters.ble.ble - INFO - GetManagedObjects
2021-08-19 15:50:56,262 - adapters.ble.ble - INFO - GetAll
2021-08-19 15:50:56,291 - adapters.ble.ble - INFO - returning props
2021-08-19 15:50:56,328 - adapters.ble.waterrowerble - INFO - GATT application registered
2021-08-19 15:50:56,346 - adapters.ble.waterrowerble - INFO - Advertisement registered
2021-08-19 15:50:57,588 - root - INFO - found SmartRow
2021-08-19 15:50:57,596 - root - INFO - de:47:16:eb:de:b6
2021-08-19 15:50:57,614 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2021-08-19 15:50:58,075 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
2021-08-19 15:51:09,480 - adapters.smartrow.smartrowreader - INFO - Connected to [de:47:16:eb:de:b6]
2021-08-19 15:51:10,063 - adapters.smartrow.smartrowreader - INFO - Disconnected [de:47:16:eb:de:b6]
Update Waterrower Data
2021-08-19 15:53:42,181 - adapters.ble.ble - INFO - Default StartNotify called, returning error
2021-08-19 15:53:42,228 - adapters.ble.waterrowerble - WARNING - no data from s4 interface`

the smartrow app works flawless so far.
any idea?

Thanks:)

Classico12, 2021-10-20:
I had a similar problem for a bit.  It seems the pirowflo starts a session when it gets data from the s4.  The same connection is required.

inonoob, 2022-03-04:
Hey if you had a version 3.0 of the smartrow which I wasn't aware exist, now it should work.

jcbigears, 2023-04-29:
I have a similar problem. Rasp pi 4B, TP-link bluetooth and ANT+ dongle. Starting app via webserver. Smartrow remains blue flashing lights. EXR running on an ipad can see Pirowflo as a device and can connect to it, but no data coming from Smartrow unit via bluetooth - just keeps flashing until timeout.

Terminal output without EXR connection below.
pi@raspberrypi:~/pirowflo/src $ python3 waterrowerthreads.py -i sr -b -a

> 2023-04-29 11:25:34,441 - __main__ - INFO - S4 not selected
> 2023-04-29 11:25:34,443 - __main__ - INFO - inferface smartrow will be used for data input
> 2023-04-29 11:25:34,444 - __main__ - INFO - Smartrow Interface started
> 2023-04-29 11:25:34,444 - __main__ - INFO - Start BLE Advertise and BLE GATT Server
> 2023-04-29 11:25:34,446 - __main__ - INFO - Start Ant and start broadcast data
> start search for dongle
> 2023-04-29 11:25:34,455 - adapters.smartrow.smartrowreader - INFO - starting discovery
> 2023-04-29 11:25:34,461 - adapters.ble.waterrowerble - INFO - Registering GATT application...
> 2023-04-29 11:25:34,475 - adapters.ble.ble - INFO - GetManagedObjects
> 2023-04-29 11:25:34,483 - adapters.ble.ble - INFO - GetAll
> 2023-04-29 11:25:34,485 - adapters.ble.ble - INFO - returning props
> 2023-04-29 11:25:34,486 - adapters.ble.waterrowerble - INFO - GATT application registered
> 2023-04-29 11:25:34,508 - adapters.ble.waterrowerble - INFO - Advertisement registered
> Using Dynastream Innovations dongle
> calibration
> set Networkkey:b'\x00\xb9\xa5!\xfb\xbdr\xc3E'
> create Channel

Webserver output:

> 2023-04-29 11:30:31,947 - __main__ - INFO - S4 not selected
> 2023-04-29 11:30:31,949 - __main__ - INFO - inferface smartrow will be used for data input
> 2023-04-29 11:30:31,950 - __main__ - INFO - Smartrow Interface started
> 2023-04-29 11:30:31,952 - __main__ - INFO - Start BLE Advertise and BLE GATT Server
> 2023-04-29 11:30:31,953 - __main__ - INFO - Start Ant and start broadcast data
> start search for dongle
> 2023-04-29 11:30:31,961 - adapters.smartrow.smartrowreader - INFO - starting discovery
> 2023-04-29 11:30:31,974 - adapters.ble.waterrowerble - INFO - Registering GATT application...
> 2023-04-29 11:30:31,992 - adapters.ble.ble - INFO - GetManagedObjects
> 2023-04-29 11:30:31,995 - adapters.ble.ble - INFO - GetAll
> 2023-04-29 11:30:31,996 - adapters.ble.ble - INFO - returning props
> 2023-04-29 11:30:31,999 - adapters.ble.waterrowerble - INFO - GATT application registered
> 2023-04-29 11:30:32,018 - adapters.ble.waterrowerble - INFO - Advertisement registered

 
[image]

So no errors as such. Webserver screenshot attached.
Any ideas?
Thanks,
John

umchemist, 2025-03-07:
I have this error as well, did anyone ever find a fix?

jcbigears, 2025-03-07:
I'm afraid I gave up