inonoob/pirowflo #43: Asensei Support

konradknitter, 2021-09-22:
Hey,

Thank you for a great work! Just found this project and it works quite well.

I was able to successfully connect it to WaterRower Connect and Kinomap iOS app.

I was checking other applications, exactly one called Asensei that is supposed to work with S4 Commodule monitor - but I’m not able to get it discovered PiRowFlo.

Was anyone successful to do so? Are we aware about differences between original Commodule and one from PiRowFlo?

inonoob, 2021-09-22:
Hey 

That could have some different reasons. One I could think of is the app is looking for the com module name and cross check with their database name. As PiRowFlo is unknow to them, the app won't see it. But what might work is to change the broadcasted name of pirowflo in the code. So it then looks like a S4 comm module. 

Maybe that helps

konradknitter, 2021-09-22:
Hi, thanks for reply. 

I change bluetooth name via

  `bluetoothctl system-alias “S4 Comms 99”`

Indeed, it helped. Thanks!

inonoob, 2021-11-05:
Fixed

closed 2021-11-05 by inonoob