inonoob/pirowflo #44: Has anybody PiRowFlo Wirkung on Forerunner 935?

ConnOnGit, 2021-10-02:
I‘m trying to use PoRowFlo with the native rowing application on my Forerunner 935 with FW 21.00.
Unfortunatelly it‘s not working. I‘m using the Garmin Ant+ USB dongle. It seems that the watch doesn‘t recognize PiRowFlo.
So the question is: is there anybody with a 935 and a working PiRowFlow using the native garmin rowing app on the watch?

inonoob, 2022-03-04:
Has the forerunner the rowing app ?

ConnOnGit, 2022-03-04:
Yes, it has the rowing app. Thats why I‘m interested to know why it isn‘t working.

inonoob, 2022-03-04:
That's so strange, I know now from 3 different people the rowing app on the Fenix 6 is working. 

The question is: do you get any connection at all with PiRowFlo but no data ?

ConnOnGit, 2022-03-04:
No- unfortunatelly the 935 doesn‘t get a ant+ connection with the pirowflo- with bluetooth there is a connection but no data is tranfered.
I do get a connection when I‘m using bluetooth from pirowflo with the waterrower app.
Perhaps there is a special thing with the 935 which is different to the fenix watches?

inonoob, 2022-03-05:
Hey 

Sorry for asking my questions:

The connection PiRowFlo via Bluetooth with the waterrower app is working but no data ? 
The waterrower app is known not working at least the last time I test it. 

Does PiRowFlo connect to the forerunner via ant+ ? What ant+ dongle do you have? 

What pi version and PiRowFlo version do you have ?

BR

inonoob, 2022-03-05:
The Fenix 6 serie got an update 12.77 which includes the following:

Added support for displaying metrics from an indoor rowing machine that supports the ANT FE-C profile (e.g. Concept 2 with the PM5 module). This includes new fields for the Indoor Rowing app: Distance, Pace, Power, and Stroke Distance.

https://www8.garmin.com/support/download_details.jsp?id=15161

I did not find it for the forerunner 945:

https://www8.garmin.com/support/download_details.jsp?id=14933

ConnOnGit, 2022-03-05:
Hi,
first of all: thank you for your support!

PiRowFlo with the waterrower is working well with bluetooth connection. The connection is working and the data are transmititted. 

PiRowFlo with the forerunner 935 is not working. I can connect PiRowFlo via bluetooth but there are no data transmitted to the rowing app.
The forerunner 935 can’t connect PiRowFlo via Ant+. When I start the discovery of sensors no PiRowFlo or sensor is found instead my other real ant+ equipment like chest strap or power meter.

I tried PiRowFlo with the newest version on raspberry pi 4 and zero wh

Perhaps the forerunner 935 does not support the ANT FE-C profile. This would explain why it doesn‘t connect the rowing app to the PiRowFlo.

inonoob, 2022-03-05:
Hey 

Sure ! Hope you still have fun with the project and you have a good training! 

You should have seen my face after hours and hours of programming just to find out that my vivoactive 3 is not working :). But at least I understood the ant+ protocol. 

Let's then hope for an update for the forerunner 

BR

inonoob, 2022-03-05:
What you could do to double check it is to see if you can have access to a concept2 with ant+. 

If the concept2 also does not connect to the forerunner945 we can confirm the app is not working with fe-c.

closed 2022-03-05 by inonoob

dan06, 2022-03-05:
Just to clear up any confusion, the FR945 works great with PiRowFlow, but I am assuming the rowing app update did not make it to the FR935.

https://www.dcrainmaker.com/2017/03/garmin-forerunner-935-depth-review.html lists the profiles about halfway down the page, and FE-C is not listed.

inonoob, 2022-03-05:
Ah you are right ! It is 935 not 945 ! Thanks for the feedback.

ConnOnGit, 2022-03-05:
Hi dan06!
Thank you for clarify this!
So the 945 is working but not the 935 because ir doesn‘t support the Ant FE-C profile.
Thanks a lot for the helpful information!

inonoob, 2022-03-05:
[image]

The list of Garmin watches with fe-c

LNS-MrGoodkat, 2023-03-11:
Hi,
I had the same issue as the OP Bluetooth to iPad works fine but no data via ant to my Venu 2.

Anyway after some research I found a App in the Garmin AppStore for the Concept 2 and it works also with PiRowFlow :).

[AK Concept2 Rowing](https://apps.garmin.com/en-US/apps/74afbe31-c6c4-47ee-8776-3b9778cb676b) 

It also works on the Edge 530.