inonoob/pirowflo #45: [BUG] ModuleNotFoundError: No module named 'gatt'

libxmike, 2021-11-13:
as far as I can tell, the installation went well, but if I try to start with:

python3 waterrowerthreads.py -i s4 -b -a 
or
python3 waterrowerthreads.py -b -a 

I get gatt error:

Traceback (most recent call last):
  File "/home/pi/pirowflo/src/waterrowerthreads.py", line 34, in <module>
    from adapters.smartrow import smartrowtobleant
  File "/home/pi/pirowflo/src/adapters/smartrow/smartrowtobleant.py", line 4, in <module>
    import gatt
ModuleNotFoundError: No module named 'gatt'

any idea what it can be?

Installed on a Rpi Zero (not W, 1st gen) with ANT+ (Dynastream_Innovations) and BT 4.0 Dongle

libxmike, 2021-11-13:
I did rerun the install script, and had same error as on the ininitial installation:

`
Building wheels for collected packages: PyGObject
  Building wheel for PyGObject (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bz7mhce5/pygobject_292c0fdbedb543d2bb483d55eb417434/setup.py'"'"'; __file__='"'"'/tmp/pip-install-bz7mhce5/pygobject_292c0fdbedb543d2bb483d55eb417434/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-nmekb6ew
       cwd: /tmp/pip-install-bz7mhce5/pygobject_292c0fdbedb543d2bb483d55eb417434/
  Complete output (51 lines):
  running bdist_wheel
`

libxmike, 2021-11-13:
Additional possible BT related issues while installing:

`
No Ant stick found !
Ant dongle found
udev rule written to /etc/udev/rules.d/99-garmin.rules
----------------------------------------------
 add user to the group bluetoot and dialout   
----------------------------------------------
 
-----------------------------------------------
 Change bluetooth name of the pi to PiRowFlo
-----------------------------------------------
 
 
------------------------------------------------------
 configuring web interface on http://rpi0-waterrower:9001 
------------------------------------------------------
 
sed: can't read services/supervisord.service: No such file or directory
sed: can't read services/supervisord.service: No such file or directory
sed: can't read services/supervisord.service: No such file or directory
mv: cannot stat 'services/supervisord.service': No such file or directory
 
------------------------------------------------------------
 Update bluetooth settings according to Apple specifications
------------------------------------------------------------
 
sed: can't read services/update-bt-cfg.service: No such file or directory
mv: cannot stat 'services/update-bt-cfg.service': No such file or directory
 
------------------------------------------------------------
 setup screen setting to start up at boot                   
------------------------------------------------------------
 
sed: can't read services/screen.service: No such file or directory
sed: can't read services/screen.service: No such file or directory
mv: cannot stat 'services/screen.service': No such file or directory
-----------------------------------------------
 update bluart file as it prevents the start of
 internal bluetooth if usb bluetooth dongle is 
 present                                       
-----------------------------------------------
----------------------------------------------
 Add absolut path to the logging.conf file    
----------------------------------------------
 
----------------------------------------------
 installation done ! rebooting in 3, 2, 1 
`

inonoob, 2021-11-13:
Hey 

Something seems not to work with pip install. So as the installation is not finished you can' t used it. 

Try to install the modules from requirements.txt manually with pip3. Then we might can see at which point the code brakes. 
I can't be that one of the module version don't work with pi zero version 1.

libxmike, 2021-11-13:
> Hey
>
> Something seems not to work with pip install. So as the installation is not finished you can' t used it.
>
> Try to install the modules from requirements.txt manually with pip3. Then we might can [...]

as I have all Rpi version to play with, will try with Rpi 3 and Rpi Zero W, will see if then the microSD works with Pi Zero 1st Gen. 
I was suspecting that may not be "compatible" with the old Zero but then asked myself, why no? :D

libxmike, 2021-11-13:
no luck with Rpi3 on clean Raspbian light

`
WARNING: The candidate selected for download or install is a yanked version: 'pyusb' candidate (version 1.1.1 at https://www.piwheels.org/simple/pyusb/pyusb-1.1.1-py3-none-any.whl#sha256=f18eb813d3a1439918071234589162c2f209a19adbeffeb1377ce078a4aebc70 (from https://www.piwheels.org/simple/pyusb/))
Reason for being yanked: <none given>
Building wheels for collected packages: PyGObject
  Building wheel for PyGObject (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-35rtgdn2/pygobject_703868eb9084482e86514691398cc6bc/setup.py'"'"'; __file__='"'"'/tmp/pip-install-35rtgdn2/pygobject_703868eb9084482e86514691398cc6bc/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-8ylqbc_u
       cwd: /tmp/pip-install-35rtgdn2/pygobject_703868eb9084482e86514691398cc6bc/
  Complete output (51 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.9
  creating build/lib.linux-armv7l-3.9/pygtkcompat
  copying pygtkcompat/__init__.py -> build/lib.linux-armv7l-3.9/pygtkcompat
  copying pygtkcompat/generictreemodel.py -> build/lib.linux-armv7l-3.9/pygtkcompat
  copying pygtkcompat/pygtkcompat.py -> build/lib.linux-armv7l-3.9/pygtkcompat
  creating build/lib.linux-armv7l-3.9/gi
  copying gi/_compat.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_signalhelper.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/types.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_error.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_propertyhelper.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/module.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_gtktemplate.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/__init__.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_option.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/pygtkcompat.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/docstring.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_constants.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/_ossighelper.py -> build/lib.linux-armv7l-3.9/gi
  copying gi/importer.py -> build/lib.linux-armv7l-3.9/gi
  creating build/lib.linux-armv7l-3.9/gi/repository
  copying gi/repository/__init__.py -> build/lib.linux-armv7l-3.9/gi/repository
  creating build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/Gdk.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/Pango.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/__init__.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/GLib.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/Gtk.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/keysyms.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/GIMarshallingTests.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/GdkPixbuf.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/Gio.py -> build/lib.linux-armv7l-3.9/gi/overrides
  copying gi/overrides/GObject.py -> build/lib.linux-armv7l-3.9/gi/overrides
  running build_ext
  pycairo: new API
  pycairo: trying include directory: '/usr/lib/python3/dist-packages/cairo'
  pycairo: header file ('/usr/lib/python3/dist-packages/cairo/py3cairo.h') not found
  pycairo: old API
  pycairo: found '/usr/lib/python3/dist-packages'
  pycairo: trying include directory: '/usr/include/pycairo'
  pycairo: header file ('/usr/include/pycairo/py3cairo.h') not found
  pycairo: pkg-config
  Package py3cairo was not found in the pkg-config search path.
  Perhaps you should add the directory containing `py3cairo.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'py3cairo' found
  Command '('pkg-config', '--print-errors', '--exists', 'py3cairo >= 1.11.1')' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for PyGObject
  Running setup.py clean for PyGObject
Failed to build PyGObject
Installing collected packages: wrapt, pyusb, pyserial, smbus2, pyftdi, pillow, deprecated, cbor2, luma.core, supervisor, PyGObject, numpy, luma.oled, gatt, dbus-python
  Attempting uninstall: PyGObject
    Found existing installation: PyGObject 3.38.0
    Not uninstalling pygobject at /usr/lib/python3/dist-packages, outside environment /usr
[... 87 lines cut ...]
 
------------------------------------------------------------
 Update bluetooth settings according to Apple specifications
------------------------------------------------------------
 
Created symlink /etc/systemd/system/multi-user.target.wants/update-bt-cfg.service → /etc/systemd/system/update-bt-cfg.service.
 
------------------------------------------------------------
 setup screen setting to start up at boot                   
------------------------------------------------------------
 
Created symlink /etc/systemd/system/multi-user.target.wants/screen.service → /etc/systemd/system/screen.service.
-----------------------------------------------
 update bluart file as it prevents the start of
 internal bluetooth if usb bluetooth dongle is 
 present                                       
-----------------------------------------------
----------------------------------------------
 Add absolut path to the logging.conf file    
----------------------------------------------
 
----------------------------------------------
 installation done ! rebooting in 3, 2, 1 
----------------------------------------------
`

libxmike, 2021-11-13:
also as of now, pirowflo can't be installed, tried on Rpi Zero 1st gen, Rpi Zero W, and Rpi 3

inonoob, 2021-11-13:
Hey 

Can you change in the requirements.tyt the pyusb version to 1.2.0 

pyusb == 1.2.0

And try again. The 1.1.1 pyusb version has been yanked or something... 

That might fix it

libxmike, 2021-11-14:
Install works after manual pip3 install of missing packages from requirements.txt 

Also needed to change the version of PyGObject in the requirements after installation, otherwise same error again.
Only numpy was downgraded as I did not change the version, but works.

`pyserial == 3.5
PyGObject == 3.38.0
dbus-python == 1.2.18
numpy == 1.20.1
pyusb == 1.2.0
gatt == 0.2.7
supervisor == 4.2.1
luma.oled == 3.8.1`

libxmike, 2021-11-14:
Just tested on my Waterrower, and it works, Rpi Zero 1st Gen works. Finely found usage for that ancient Pi and old BT/WiFi dongles :D

inonoob, 2021-11-14:
Thanks for the update here ! 

I've done the changes in the master. So it should be fixed 

BR

Ino

closed 2021-11-14 by inonoob