inonoob/pirowflo #46: Issue with pycairo during install

27escape, 2021-11-14:
Hi, not sure if pycairo is needed or not but it throws up errors during the install process

```

----------------------------------------------
install needed python3 modules for the project
----------------------------------------------

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyserial==3.5 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 1)) (3.5)
Collecting PyGObject==3.30.4
  Using cached PyGObject-3.30.4.tar.gz (690 kB)
Collecting dbus-python==1.2.16
  Using cached https://www.piwheels.org/simple/dbus-python/dbus_python-1.2.16-cp39-cp39-linux_armv6l.whl (374 kB)
Collecting numpy==1.20.1
  Using cached https://www.piwheels.org/simple/numpy/numpy-1.20.1-cp39-cp39-linux_armv6l.whl (11.6 MB)
Requirement already satisfied: pyusb==1.1.1 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 5)) (1.1.1)
Collecting gatt==0.2.7
  Using cached https://www.piwheels.org/simple/gatt/gatt-0.2.7-py3-none-any.whl (9.7 kB)
Requirement already satisfied: supervisor==4.2.1 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 7)) (4.2.1)
Collecting luma.oled==3.8.1
  Using cached https://www.piwheels.org/simple/luma-oled/luma.oled-3.8.1-py2.py3-none-any.whl (32 kB)
Requirement already satisfied: luma.core>=2.0.0 in /usr/local/lib/python3.9/dist-packages (from luma.oled==3.8.1->-r requirements.txt (line 8)) (2.3.1)
Requirement already satisfied: pycairo>=1.11.1 in /usr/lib/python3/dist-packages (from PyGObject==3.30.4->-r requirements.txt (line 2)) (1.16.2)
Requirement already satisfied: pyftdi in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (0.53.3)
Requirement already satisfied: deprecated in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (1.2.13)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (3.5)
Requirement already satisfied: cbor2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (5.4.2)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (0.4.1)
Requirement already satisfied: RPI.GPIO in /usr/lib/python3/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (0.7.0)
Requirement already satisfied: pillow>=4.0.0 in /usr/local/lib/python3.9/dist-packages (from luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (8.4.0)
Requirement already satisfied: wrapt<2,>=1.10 in /usr/local/lib/python3.9/dist-packages (from deprecated->luma.core>=2.0.0->luma.oled==3.8.1->-r requirements.txt (line 8)) (1.13.3)
Building wheels for collected packages: PyGObject
  Building wheel for PyGObject (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6qdvqqwj/pygobject_14bf217bee7447f88ff425f8eed348e5/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6qdvqqwj/pygobject_14bf217bee7447f88ff425f8eed348e5/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-xuihy309
       cwd: /tmp/pip-install-6qdvqqwj/pygobject_14bf217bee7447f88ff425f8eed348e5/
  Complete output (51 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv6l-3.9
  creating build/lib.linux-armv6l-3.9/pygtkcompat
  copying pygtkcompat/__init__.py -> build/lib.linux-armv6l-3.9/pygtkcompat
  copying pygtkcompat/generictreemodel.py -> build/lib.linux-armv6l-3.9/pygtkcompat
  copying pygtkcompat/pygtkcompat.py -> build/lib.linux-armv6l-3.9/pygtkcompat
  creating build/lib.linux-armv6l-3.9/gi
  copying gi/_compat.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_signalhelper.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/types.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_error.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_propertyhelper.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/module.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_gtktemplate.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/__init__.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_option.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/pygtkcompat.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/docstring.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_constants.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/_ossighelper.py -> build/lib.linux-armv6l-3.9/gi
  copying gi/importer.py -> build/lib.linux-armv6l-3.9/gi
  creating build/lib.linux-armv6l-3.9/gi/repository
  copying gi/repository/__init__.py -> build/lib.linux-armv6l-3.9/gi/repository
  creating build/lib.linux-armv6l-3.9/gi/overrides
  copying gi/overrides/Gdk.py -> build/lib.linux-armv6l-3.9/gi/overrides
  copying gi/overrides/Pango.py -> build/lib.linux-armv6l-3.9/gi/overrides
  copying gi/overrides/__init__.py -> build/lib.linux-armv6l-3.9/gi/overrides
  copying gi/overrides/GLib.py -> build/lib.linux-armv6l-3.9/gi/overrides
  copying gi/overrides/Gtk.py -> build/lib.linux-armv6l-3.9/gi/overrides
[... 114 lines cut ...]
------------------------------------------------------------

sed: can't read services/update-bt-cfg.service: No such file or directory
mv: cannot stat 'services/update-bt-cfg.service': No such file or directory

------------------------------------------------------------
 setup screen setting to start up at boot
------------------------------------------------------------

sed: can't read services/screen.service: No such file or directory
sed: can't read services/screen.service: No such file or directory
mv: cannot stat 'services/screen.service': No such file or directory
-----------------------------------------------
 update bluart file as it prevents the start of
 internal bluetooth if usb bluetooth dongle is
 present
-----------------------------------------------
```

This is on a Rasberry Pi Zero W version 1.2 running the latest Raspberry Pi OS Lite, released October 30th 2021.

This is my first try with piroflow, so I want to make sure that I have everything right!

Many thanks for creating this project
Kevin

inonoob, 2021-11-14:
Hey Kevin,

please retry. I've updated all the requirements versions 

Please give me a feedback if it is working now.

BR 

Ino

27escape, 2021-11-14:
Many Thanks Ino, Thats fixed it - thanks for the quick response!
Now to find a suitable cable to connect my S4 monitor to my Pi and I should be good to go.

Kevin

inonoob, 2021-11-14:
Fixed updated requirements.txt

closed 2021-11-14 by inonoob