inonoob/pirowflo #53: Kinomap working with SmartRow 3.0 on IOS

DANgerous25, 2022-03-03:
Hi, thanks to your latest work with @dan06 my PiRoFlow is working with my new SmartRow. I notice on your homepage it says Kinomap is "Pending Test". I've done a very quick test after getting everything working, it appears to work in that Kinomap gets the SmartRow values and these are displayed continually in the app.

My question, and I'm not sure if this is a Kinomap issue, a non-issue, user error or what... but is the SmartRow supposed to control the video speed of rowing in Kinomap? Because mine doesn't seem to, even though I've selected "at my speed". For example, if I take a few strokes then stop rowing, the video keeps playing for a while even though no SmartRow data is being transmitted/received.

Perhaps my view of what Kinomap is/does is wrong. Sorry if this is a dumb question.

inonoob, 2022-03-03:
Hey

you are welcome. Hope you have fun and can enjoy the full potential of the smartrow.

One question: For which OS did it work apple or android ?

DANgerous25, 2022-03-04:
IOS. Thanks a lot.

inonoob, 2022-03-04:
Thanks for your feedback !

inonoob, 2022-03-04:
Will update the readme

closed 2022-03-04 by inonoob