inonoob/pirowflo #54: [BUG] Distance and Pace metric are not shown on Garmin device / Garmin Connect

stampede79, 2022-03-04:
Hi,

first of all thx for your effort concerning the chnages made for smartrow 3.0!

unfortunately the metrics for distance and avg speed (perhaps some more) are neither shown on Garmin Fenix 6 (indoor-rowing-app) nor later in the exported training on garmin connect. I have updated your pirowflo to the latest version. On Coxswain all metrics are shown correctly.

Do you have any idea what may be the issue ?

Regards 
Jan

inonoob, 2022-03-04:
Hey,

Did you upgrade from a working version ? Do you have a SmartRow 3.0 or 2.0? 

@Dan has test it on his Garmin watch and he posted some screenshots. 

I don't own a Fenix 6. 

Does it work for the S4 computer ? 

Coxswain is working via Bluetooth. Does the watch connect to the pi ?

stampede79, 2022-03-04:
Hey,
thx for answering so fast!
i did the upgrade from a running version - correct.
how do i figure out which smartrow version I own ?
my watch connects to the pi.
i have no s4 computer.

Regards
Jan

inonoob, 2022-03-04:
Hey, 

The update broke your working version ?

inonoob, 2022-03-04:
And coxswain is show the correct numbers ?

stampede79, 2022-03-04:
Seems so as if.
Last Training which worked without problems was on Feb 18th. I havent trained for some time. Today, after the update of pirowflo, metrics were shown at coxswain but not on my fenix and in garmin connect. (after upload)

stampede79, 2022-03-04:
Metric like "rate" is shown in Garmin Connect and the watch.

inonoob, 2022-03-04:
The thing is Bluetooth and Ant+ get the same data. I'm a bit worried that the distance decoding has broke something. 

So I'm surprised that coxswain show the distance but ant+ does not.

inonoob, 2022-03-04:
Ok so that means the watch a get some data. Just distance and distance per time unit is not working.

stampede79, 2022-03-04:
I will not exclude that i did sth. wrong - perhaps i have to reinitialize the ant+ dongle ?!

yes - some data (from pirowflo and the pulsestrap) are shown on the watch...(i.e current stroke rate)

inonoob, 2022-03-04:
I start to think that the distance has a bug due to smartrow version 3. 

Can you double check again if the coxswain value are correct ? I'm really suspecting the distance value.

stampede79, 2022-03-04:
Coxswain values seem to be plausible. (4400m in 20 min (values as always during my training))

Just to illustrate my statement:
[image]

inonoob, 2022-03-04:
Ok,
I will it up and retry to check the ant+ part. I don't own a Fenix 7 but I can debut via Android app and check the values. 

Had you try to unpair and repair with the pi ?

stampede79, 2022-03-04:
Yes I did newly pair the devices. I had a look in the log and saw, that i use a CYCPLUS dongle which you do not recommend (which worked fine before). Perhaps I change that first and give you a feedback...Sorry for the inconveniance if that is the root cause...

inonoob, 2022-03-04:
Hey no problem,

I'm going to test it with my Garmin dongle. 

https://github.com/WouterJD/FortiusANT/issues/61

Cycplus was bad also for me and had strange behaviour. Sometime working and sometimes not

stampede79, 2022-03-04:
Mea Culpa!!!  Seems to be an ANT+ Dongle Issue. The other one sends the data properly and my garmin device receives all data. It worked fine before - don't know why. Thx again for your effort and the whole project!!!
Regards Jan

inonoob, 2022-03-04:
Hey 

Very nice so happy again !! :) 

Very nice. You are welcome have fun with the project ! 

BR

inonoob, 2022-03-04:
Those cheap ant+ dongle are a pain de debug as you may have seen from the other project FortibusAnt. They even try to fix it in their code but mentioned it is not fully working.

closed 2022-03-04 by inonoob