inonoob/pirowflo #55: [BUG] Something is not working

DANgerous25, 2022-03-12:
**Describe the bug**
When using SmartRow [only, no S4], PiRowFlo only occasionally shows that it is receiving SmartRow data. I think it's something to do with the startup sequence and connectivity of devices, but I haven't been able to nail a pattern. It rarely works for me, and if it doesn't work there's no way I can get it to work unless I disconnect the SmartRow battery and try again a few times - even rebooting the Pi doesn't help. However on the odd occasion it does work, and keeps working until I switch it off.

I don't know how to debug what is happening, can you please tell me how to debug from the logs, and tell me the correct sequence for startup? And if it doesn't work connect, what's the correct sequence to reset everything? 

**To Reproduce**
Turn on Pi
Start both PiRowFlo or SmartRow in any sequence
Observe initial STDOUT messages from PiRowFlo
Start Rowing
Observe that nothing appears in STDOUT

**Expected behavior**

Turn on Pi
Start both PiRowFlo or SmartRow in any sequence
Observe initial STDOUT messages from PiRowFlo
Start Rowing
Observe SmartRow data in STDOUT

**Screenshots**
Example of it doing nothing:

pi@pi1:~/pirowflo/src $ python3 waterrowerthreads.py -i sr -b
2022-03-12 19:00:50,097 - __main__ - INFO - S4 not selected
2022-03-12 19:00:50,098 - __main__ - INFO - inferface smartrow will be used for data input
2022-03-12 19:00:50,098 - __main__ - INFO - Smartrow Interface started
2022-03-12 19:00:50,099 - __main__ - INFO - Start BLE Advertise and BLE GATT Server
2022-03-12 19:00:50,099 - __main__ - INFO - Ant service not used
2022-03-12 19:00:50,107 - adapters.smartrow.smartrowreader - INFO - starting discovery
2022-03-12 19:00:50,116 - adapters.ble.waterrowerble - INFO - Registering GATT application...
2022-03-12 19:00:50,123 - adapters.ble.ble - INFO - GetManagedObjects
2022-03-12 19:00:50,125 - adapters.ble.ble - INFO - GetAll
2022-03-12 19:00:50,126 - adapters.ble.ble - INFO - returning props
2022-03-12 19:00:50,129 - adapters.ble.waterrowerble - INFO - GATT application registered
2022-03-12 19:00:50,158 - adapters.ble.waterrowerble - INFO - Advertisement registered

**Smartphone (please complete the following information):**
Raspberry Pi 4 with additional BlueTooth dongle
Watterrower Performance Ergometer (only has SmartRow, no S4)

**Additional context**
Add any other context about the problem here.

inonoob, 2022-03-12:
Hey

Can you check which version of bluez is installed on your pi ? 

Bluez --version

You could play with the delay in the code. 

https://github.com/inonoob/pirowflo/blob/6ea5f3a9d224ed594b23c25c186737bc0cae7ac3/src/adapters/smartrow/smartrowtobleant.py#L266

inonoob, 2022-03-12:
Does your log look like this: 

2021-07-12 12:00:36,147 - main - INFO - S4 not selected
2021-07-12 12:00:36,149 - main - INFO - inferface smartrow will be used for data input
2021-07-12 12:00:36,159 - main - INFO - Smartrow Interface started
2021-07-12 12:00:36,160 - main - INFO - Start BLE Advertise and BLE GATT Server
2021-07-12 12:00:36,163 - main - INFO - Start Ant and start broadcast data
2021-07-12 12:00:36,261 - adapters.smartrow.smartrowreader - INFO - starting discovery
2021-07-12 12:00:36,281 - adapters.ble.waterrowerble - INFO - Registering GATT application...
2021-07-12 12:00:36,294 - adapters.ble.ble - INFO - GetManagedObjects
2021-07-12 12:00:36,299 - adapters.ble.ble - INFO - GetAll
2021-07-12 12:00:36,300 - adapters.ble.ble - INFO - returning props
2021-07-12 12:00:36,309 - adapters.ble.waterrowerble - INFO - GATT application registered
2021-07-12 12:00:36,396 - adapters.ble.waterrowerble - INFO - Advertisement registered
2021-07-12 12:00:39,136 - root - INFO - found SmartRow

The "2021-07-12 12:00:39,136 - root - INFO - found SmartRow" or what is been stdout after the "2021-07-12 12:00:36,396 - adapters.ble.waterrowerble - INFO - Advertisement registered" ?

BR

inonoob, 2022-03-12:
This is what it should looks like after your STDOUT: 

2021-07-12 12:00:36,396 - adapters.ble.waterrowerble - INFO - Advertisement registered
2021-07-12 12:00:39,136 - root - INFO - found SmartRow
2021-07-12 12:00:39,138 - root - INFO - cfded9:8d:3b:75
2021-07-12 12:00:39,140 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2021-07-12 12:00:39,221 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
2021-07-12 12:00:40,807 - adapters.smartrow.smartrowreader - INFO - Connected to [XX:XX:XX:XX:XX] <== your SmartRow mac address. 

BR

DANgerous25, 2022-03-12:
THanks for your responses. This is what I get when it DOES work. When it doesn't work, all I get was the output from my previous post.

btw. "Bluez --version" command doesn't work, says command not found. But if I do bluetoothd -v it is 5.5.

pi@pi1:~/pirowflo/src $ python3 waterrowerthreads.py -i sr -b
2022-03-12 20:16:38,676 - __main__ - INFO - S4 not selected
2022-03-12 20:16:38,676 - __main__ - INFO - inferface smartrow will be used for data input
2022-03-12 20:16:38,677 - __main__ - INFO - Smartrow Interface started
2022-03-12 20:16:38,677 - __main__ - INFO - Start BLE Advertise and BLE GATT Server
2022-03-12 20:16:38,677 - __main__ - INFO - Ant service not used
2022-03-12 20:16:38,687 - adapters.smartrow.smartrowreader - INFO - starting discovery
2022-03-12 20:16:38,692 - adapters.ble.waterrowerble - INFO - Registering GATT application...
2022-03-12 20:16:38,697 - adapters.ble.ble - INFO - GetManagedObjects
2022-03-12 20:16:38,699 - adapters.ble.ble - INFO - GetAll
2022-03-12 20:16:38,700 - adapters.ble.ble - INFO - returning props
2022-03-12 20:16:38,703 - adapters.ble.waterrowerble - INFO - GATT application registered
2022-03-12 20:16:38,714 - adapters.ble.waterrowerble - INFO - Advertisement registered
2022-03-12 20:16:45,080 - root - INFO - found SmartRow
2022-03-12 20:16:45,081 - root - INFO - d8:be:21:9a:e4:6f
2022-03-12 20:16:45,083 - adapters.smartrow.smartrowreader - INFO - found SmartRow macaddress
2022-03-12 20:16:45,186 - adapters.smartrow.smartrowtobleant - INFO - SmartRow Ready and sending data to BLE and ANT Thread
2022-03-12 20:16:45,681 - adapters.smartrow.smartrowreader - INFO - Connected to [d8:be:21:9a:e4:6f]
2022-03-12 20:16:46,458 - adapters.smartrow.smartrowreader - INFO - Resolved services [d8:be:21:9a:e4:6f]
2022-03-12 20:16:46,461 - adapters.smartrow.smartrowreader - INFO - 	[d8:be:21:9a:e4:6f] Service [00001234-0000-1000-8000-00805f9b34fb]
2022-03-12 20:16:46,461 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001236-0000-1000-8000-00805f9b34fb]
2022-03-12 20:16:46,462 - adapters.smartrow.smartrowreader - INFO - 		Characteristic [00001235-0000-1000-8000-00805f9b34fb]
2022-03-12 20:16:46,463 - adapters.smartrow.smartrowreader - INFO - 	[d8:be:21:9a:e4:6f] Service [00001801-0000-1000-8000-00805f9b34fb]
starting heart beat

inonoob, 2022-03-12:
Can you try with a second Bluetooth dongle ? 

You could play with timing in either here: 

https://github.com/inonoob/pirowflo/blob/6ea5f3a9d224ed594b23c25c186737bc0cae7ac3/src/adapters/smartrow/smartrowreader.py#L120

or here

add a sleep(0.2) here: 

https://github.com/inonoob/pirowflo/blob/6ea5f3a9d224ed594b23c25c186737bc0cae7ac3/src/waterrowerthreads.py#L105

This might help. I can't debug on the unit right now. 

BR

DANgerous25, 2022-03-12:
Thanks a lot, I will have a play and will try to debug.

DANgerous25, 2022-03-14:
I don't have the full picture yet, but I can now get it to connect reasonably reliably by restarting the bluetooth service, ensuring that SmartRow is not paired to my IOS device, then starting the start script. Perhaps it is worth putting a wrapper script around the start script, and restarting bluetooth before starting PiRowFlo?

`pi@pi1:~/pirowflo/src $ sudo systemctl restart bluetooth.service
pi@pi1:~/pirowflo/src $ python3 waterrowerthreads.py -i sr -b`

Hence I think this can be closed for now, although I have some other problems with my device retaining working bluetooth connectivity for use in Kinomap and EXR. (I will raise a separate issue for both if necessary after more testing)

closed 2022-03-14 by DANgerous25