inonoob/pirowflo #63: Waterrower Connect broken compatibility

JPOvenden, 2023-01-10:
Wondering if a recent update for WR Connect means it will no longer recognise PiRowFlo?

jamespilgrim, 2023-01-10:
I just tried it with version 2.13.3 as of 1/1/23, and it does not appear to work.  It does work for things like EXR, though.  I just started working with pirowflo (and thank you all for creating it!! :), so I haven't yet figured out how to enable debugging/logging to find out more about what's happening.  I don't see any errors from the Pi console, but RW Connect just doesn't find the device.

I'd be happy to try to figure out what's going on (just retired so I've finally got time to have some fun!).  Could anyone point me in the direction of any useful information about the connect protocol and/or instrumentation to help track the back-and-forth between the app and the program?  I'm sure it's probably apparent from the code, but if there's an advanced starting position that someone else used, I'd appreciate it.

inonoob, 2023-01-10:
Hey all, 

I haven't done stuff in a long time and currently don't have the time anymore. But from my experience during my very active time, I had the issue with the App Coxwain. Where I could make a connection to the app but just after 1 sec the connection has been terminated. 

I remember that Coxwain had the Bluetooth implementation for Android not 100% correct. Sven had mad a change so that the connection was stable. I don't know if the Waterrower Connect has also the same problem. 

Another point could be that Bluez (bluetooth system for linux) set the bluetooth hardware correctly to BLE. This might also make a difference. I think sometimes Bluez does a mixture of BL + BLE or something. 
If forced to be BLE only it might help to have a connection with the App. 

BR

JPOvenden, 2023-01-10:
I had it working great before, have had to revert to the abysmal ComModule temporarily.

The Pi as PiRowFlo connected over Bluetooth to my other devices just fine and in fact an out of data version of WaterRower Connect on my iPad happily recorded a session (don't do this, your workout will not sync with your MaterRower profile or any connected apps correctly).

I might get a chance to investigate further but wanted to see if anyone else was having issues.

Yann-F, 2023-01-21:
Yes go the same issue with smartrow. The apps on Android as well as on IOs doesn’t find anymore piRowFlo.piRowFlo connect correctly to the smartrow and the garmin FR 255.

nickbits1024, 2023-01-29:
I have a similar project to this and it also quit working with WaterRower Connect.  However, I no longer have the ComModule to debug and fix the issue. If someone can use nRF Connect to connect the ComModule and post screenshots of the advertising and client tab I maybe be able to fix PiRowFlo as well as my project.  I suspect that it is being picky about the device name or some previously unnecessary service guids to make it work.

john-2000, 2023-07-27:
Hi,
I am wondering on building one of these does anyone have a fix? what setup is currently working for them?
thanks