inonoob/pirowflo #67: [BUG] OLED display not working

benbamm, 2023-04-27:
I have installed pirowflo on a rapsberry, and it is working, but the OLED display does not seem to work. It lights up entirely and only blinks every few seconds. Do I have to install something else to get the display to work or did something go wrong with the installation?

jcbigears, 2023-05-19:
Hey I have the same problem. Installed 1.3inch OLED HAT and installed from here https://www.waveshare.com/wiki/1.3inch_OLED_HAT
Screen works with python demo program.
Piroflo can be started via webserver. 
BUT screen OLED HAT remains blank.

ssfpr, 2023-08-13:
Mine does not seem to even light up. Have you found a way to make the screen work?

jcbigears, 2023-08-13:
I emailed the supplier of the screen and they sent me an image to run the
pi. Then it worked. But I gave up in the end.
John
***@***.***

<https://github.com/inonoob/pirowflo/issues/67#issuecomment-1676415997>, or
unsubscribe

benbamm, 2023-08-14:
Yeah, I did not look in to it further either. In spring and summer I don't row, so I did not use the waterrower anymore since. I could get it to work with the browser, but the screen did nothing. But I mainly built it to register the workout on my Garmin Edge 530 but this did not work, so money down the drain...

ssfpr, 2023-08-14:
I have managed to use it via the browser but the screen shows no signs of life. Am picking up the Ant signal as a Trainer on an Epix watch and it registers all the stats from the WR unit. Will see if the supplier can help.

ssfpr, 2023-08-14:
Any idea btw whether the interface used is SPI or I2C? Do I enable both or only one of them?

pc1246, 2024-04-05:
Hi guys
As you have PiRowFlo up and running, I would be interested how you have managed this.
I tried a lot, but I ended up as the others in an incomplete installation.
Do you see a chance to describe your setup?
Thanks in advance and regards
Christoph

jcbigears, 2024-04-05:
Hi. Unfortunately mine never completely worked. The pi worked fine in the end but I could never get it to connect to the Smartrow. John

ssfpr, 2024-04-07:
Mine works other than the useless screen with controller which I think is a hardware problem potentially. I turn it on via mobile browser with IP address after plugging in and I only use it to feed the signal to my Garmin Epix2. 

 

Let me know how I can help.