inonoob/pirowflo #70: Problem starting .py script and webserver on Pi4/OS Bookworm

ILAHWWINC, 2023-12-13:
I try to install pirowflo on a Pi4 / RaspberryOS Bookworm Full but it will not run.
I can not connexct to the webserver. When i run the script from commadline i get this output.

pi@raspberrypi:~/pirowflo/src $ python3 waterrowerthreads.py -i s4 -b -a
Traceback (most recent call last):
  File "/home/pi/pirowflo/src/waterrowerthreads.py", line 34, in <module>
    from adapters.smartrow import smartrowtobleant
  File "/home/pi/pirowflo/src/adapters/smartrow/smartrowtobleant.py", line 4, in <module>
    import gatt
ModuleNotFoundError: No module named 'gatt'

TbsJah, 2024-06-27:
you have to do some steps

`sudo apt-get install -y python3 python3-gi python3-dev python3-gi-cairo gir1.2-gtk-3.0 python3-pip 
sudo apt-get install -y libatlas-base-dev libglib2.0-dev libgirepository1.0-dev libcairo2-dev zlib1g-dev 
sudo apt-get install -y libfreetype6-dev liblcms2-dev libopenjp2-7 libtiff6
sudo apt-get install -y build-essential libdbus-glib-1-dev 
sudo apt-get install -y git virtualenv`

And run the installation with venv
`python3 -m venv pirowflo
source pirowflo/bin/activate

pip3 install pyserial
pip3 install PyGObject
pip3 install dbus-python
# pip3 install numpy
pip3 install pyusb
pip3 install gatt 
pip3 install supervisor 
pip3 install luma.oled
pip3 install spidev`

squiggs1982, 2024-08-10:
Hi

Thanks to @TbsJah for the above. I had the same issue as @ILAHWWINC and this has moved me forward, but I still can't quite get it over the line. 

When I run pirowflo in the venv (which I'm assuming I'm supposed to do), I get the following:

(pirowflo) chocolate@raspberrypi:~/pirowflo/src $ python3 waterrowerthreads.py
Traceback (most recent call last):
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/_core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/_core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/_core/overrides.py", line 8, in <module>
    from numpy._core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/__init__.py", line 114, in <module>
    from numpy.__config__ import show as show_config
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy._core._multiarray_umath import (
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/_core/__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/chocolate/pirowflo/src/pirowflo/bin/python3"
  * The NumPy version is: "2.0.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/chocolate/pirowflo/src/waterrowerthreads.py", line 32, in <module>
    from adapters.s4 import wrtobleant
  File "/home/chocolate/pirowflo/src/adapters/s4/wrtobleant.py", line 10, in <module>
    import numpy
  File "/home/chocolate/pirowflo/src/pirowflo/lib/python3.11/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

If I run from outside the venv, instead, it still can't find gatt. 

I've also tried running without the "python" command (so just './waterrowerthreads.py -i s4 -b -a', but that gives me a slightly different error:

Python script to broadcast waterrower data over BLE and ANT

      PiRowFlo for Waterrower
                                                                 +-+
                                               XX+-----------------+
                  +-------+                 XXXX    |----|       | |
                   +-----+                XXX +----------------+ | |
                   |     |             XXX    |XXXXXXXXXXXXXXXX| | |
    +--------------X-----X----------+XXX+------------------------+-+
    |                                                              |
    +--------------------------------------------------------------+

To begin choose an interface from where the data will be taken from either the S4 Monitor connected via USB or
the Smartrow pulley via bluetooth low energy

Then select which broadcast methode will be used. Bluetooth low energy or Ant+ or both.

e.g. use the S4 connected via USB and broadcast data over bluetooth and Ant+

python3 waterrowerthreads.py -i s4 -b -a
: File name too long
./waterrowerthreads.py: 24: import: not found
./waterrowerthreads.py: 25: import: not found
./waterrowerthreads.py: 26: import: not found
./waterrowerthreads.py: 27: import: not found
./waterrowerthreads.py: 28: from: not found
./waterrowerthreads.py: 29: from: not found
./waterrowerthreads.py: 31: from: not found
./waterrowerthreads.py: 32: from: not found
./waterrowerthreads.py: 33: from: not found
./waterrowerthreads.py: 34: from: not found
./waterrowerthreads.py: 35: import: not found
./waterrowerthreads.py: 36: import: not found
./waterrowerthreads.py: 38: Syntax error: "(" unexpected

Thanks in advance!

TbsJah, 2024-08-11:
Try to install the project from here
https://github.com/TbsJah/pirowflo

It also took me a while to find the errors. 
The cause was the wrong dependencies and venv

squiggs1982, 2024-08-11:
> Try to install the project from here https://github.com/TbsJah/pirowflo
> 
> It also took me a while to find the errors. The cause was the wrong dependencies and venv

Thank you so much @TbsJah! This worked like a charm (after I realised I needed to activate the venv and launch from there)! I now have full data from my smartrow (including power readings, stroke rate etc) going to my Garmin Epix and EXR perfectly! I am also unreasonably excited by the whole thing! :-)

If you and @inonoob have coffee pages, happy to buy you one. In any event, very much appreciate your hard work. Thank you again!

TbsJah, 2024-08-12:
All the credit goes to @inonoob  and of course to @dan06!!
I have done nothing more than summarise what others have already done :)

inonoob, 2024-08-12:
Dear all, 

I'm sorry, that I haven't manage the project in a long time. I still row very regularly but life caught up with me. Meaning, my time for this project is  nearly 0. And I must admit my programming skills are no longer on the level it was at the beginning of the project. 

So If we have merge request so we can make the project work again without too much trouble please. Let me know. 

Glad that I could create something that helps people. 

Enjoy

Einkatzi, 2024-08-31:
> > Try to install the project from here https://github.com/TbsJah/pirowflo
> > It also took me a while to find the errors. The cause was the wrong dependencies and venv
> 
> Thank you so much @TbsJah! This worked like a charm (after I realised I needed to activate the venv and launch from there)!

Could one of you (@TbsJah or @squiggs1982 ) please tell me what I need to do to activate the venv in this context and use it?
I haven't managed that yet, I guess that's why I'm running into this error:
```
pi@pi:~/pirowflo/src $ python3 waterrowerthreads.py -i s4 -b
Traceback (most recent call last):
  File "/home/pirowflo/pirowflo/src/waterrowerthreads.py", line 31, in <module>
    from adapters.ble import waterrowerble
  File "/home/pirowflo/pirowflo/src/adapters/ble/waterrowerble.py", line 10, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'
```

TbsJah, 2024-08-31:
Install as descriped here
https://github.com/TbsJah/pirowflo/blob/master/install.sh
...
`sudo apt-get install -y python3 python3-gi python3-dev python3-gi-cairo gir1.2-gtk-3.0 python3-pip 
sudo apt-get install -y libatlas-base-dev libglib2.0-dev libgirepository1.0-dev libcairo2-dev zlib1g-dev 
sudo apt-get install -y libfreetype6-dev liblcms2-dev libopenjp2-7 libtiff6
sudo apt-get install -y build-essential libdbus-glib-1-dev 
sudo apt-get install -y git virtualenv

echo " "

echo " "
echo "----------------------------------------------"
echo "install needed python3 modules for the project        "
echo "----------------------------------------------"
echo " "
python3 -m venv pirowflo
source pirowflo/bin/activate

pip3 install pyserial
pip3 install PyGObject
pip3 install dbus-python
# pip3 install numpy
pip3 install pyusb
pip3 install gatt 
pip3 install supervisor 
pip3 install luma.oled`
...

Or just install direct from here :)
sudo apt-get install git
git clone https://github.com/TbsJah/pirowflo.git
cd pirowflo 
sudo chmod +x install.sh
sudo ./install.sh 

And again
All the credit goes to @inonoob and of course to @dan06