inonoob/pirowflo #71: [BUG] Installation failing

c0zm0, 2024-01-03:
Hi,

just tried the installation on a fresh raspi zero w with bullseye (32bit).

Building wheels for collected packages: pillow
  Building wheel for pillow (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmp_octye1p_in_process.py build_wheel /tmp/tmpy8rguft7
       cwd: /tmp/pip-install-bh38l34k/pillow_efbb9bc5e7cc484bb15ccdfd966e4f9d
  Complete output (203 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv6l-cpython-39
  creating build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PdfImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/XbmImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/MicImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/FontFile.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PcxImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageMode.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/_util.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/FtexImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/__init__.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageColor.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/WalImageFile.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageOps.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageEnhance.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/TarIO.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageFont.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PixarImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageFilter.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/GimpPaletteFile.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageWin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/JpegPresets.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PdfParser.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageSequence.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageCms.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageTk.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/GdImageFile.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PpmImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/_deprecate.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImtImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PSDraw.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ContainerIO.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/FpxImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageMath.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImageStat.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/GifImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/ImagePath.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/MpoImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/FitsImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/PalmImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/BufrStubImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/TiffTags.py -> build/lib.linux-armv6l-cpython-39/PIL
  copying src/PIL/MpegImagePlugin.py -> build/lib.linux-armv6l-cpython-39/PIL
[... 158 lines cut ...]
 Change bluetooth name of the pi to PiRowFlo
-----------------------------------------------

------------------------------------------------------
 configuring web interface on http://pirowflo:9001
------------------------------------------------------

rm: cannot remove '/tmp/pirowflo*': No such file or directory
rm: cannot remove '/tmp/supervisord.log': No such file or directory

I had an ANT+ stick present at installation.

Running piroflo after rebooot fails:

~/pirowflo/src $ python3 waterrowerthreads.py -h
Traceback (most recent call last):
  File "/home/pi/pirowflo/src/waterrowerthreads.py", line 33, in <module>
    from adapters.ant import waterrowerant
  File "/home/pi/pirowflo/src/adapters/ant/waterrowerant.py", line 4, in <module>
    from . import antdongle as ant
  File "/home/pi/pirowflo/src/adapters/ant/antdongle.py", line 12, in <module>
    import usb.core
ModuleNotFoundError: No module named 'usb'

Is there a chance that this project will get an update/fix in the near future?

pc1246, 2024-04-01:
Hi
Did you fix this problem? I would like to use this software as well!
Regards Christoph

olaims, 2025-02-14:
same problem

elp3dr0, 2025-03-31:
The latest code hopefully fixes installation issues. I don't have a zero, but it works on a Pi 3. Please try again and report back so that the bug report be closed. Thanks!

inonoob, 2025-03-31:
@elp3dr0 thank you for fixing the code and updating the repo !

oldgitdaddy, 2025-06-06:
Hi @elp3dr0,
The logs are the same as from the original author of this issue. I run a raspi zero 2W with the latest raspbian OS.
```
Operating System: Raspbian GNU/Linux 12 (bookworm)
Kernel: Linux 6.12.25+rpt-rpi-v6

```

For the time being I remove the following line from requirements.txt:
`luma.oled==3.14.0`
With this the installation works, as luma oled is included for rendering of the on computer display, which the zero does not have.

I will let you know, if I can get it to work on the rower, the webserver for configuring at least is already starting.

elp3dr0, 2025-06-06:
In this case, the luma.oled package is used for the little oled screen which inonoob attached to the GPIO pins. As both the RPi3 and Zero support GPIO, I'm surprised that the Zero behaves differently. 
#Potential Cause#
 Taking a closer look at the OP's wheel logs it seems like the zero is trying to build wheel from source, and is getting stuck because it's is missing JPEG development libraries. (which presumably are needed for the oled package?) 
#Potential Fix#
Google's Gemini suggests sudo apt install libjpeg-dev as a potential system package (apt not pip) to provide the required development libraries for JPEG.

If, however, you decide simply to remove luma.oled, you could experiment with commenting out or removing the 'setup screen setting to start up at boot ' section from the install script. That configures a systemd service to start at boot that will likely need the luma.oled package and so might fail. Hopefully it won't affect the main routine though because at first glance it is just a background service.