inonoob/pirowflo #80: Question: What part of the code is connecting as a client to a BLE heart rate monitor?

elp3dr0, 2025-04-04:
I'd be grateful if someone could point me towards the part of the code that scans for and connects to the blue tooth heart rate monitor (i.e. RPi acting as a client). Ideally the actual function or lines please.

inonoob, 2025-04-04:
Quick question why should pirowflo connect to a for example polar heart rate monitor ? 

Do you mean you want the hear rate sensor connect via Bluetooth to pirowflow and combine the data from the rower with the heart rate data ? 

If yes, I think someone did implement it but I haven't got the PR from his repo

elp3dr0, 2025-04-04:
Ah! I've got myself confused from looking at too many projects.

Yes, I'm interested in combining data from a heart rate monitor connected to the RPi via Bluetooth with the data from the S4 and then transmitting the combination. I thought it was part of PiRowFlo. My mistake.

Thanks for the quick response - that's saved losing more time trying to dig through the code for something that's not there!

inonoob, 2025-04-04:
Hey,

I used to be on my agenda but then life hit and therefore I need to a bit move one. I hadn't too much time for the project. Thanks for updating the code. 

I my can check who did add the part of the code. Maybe someone can post a PR to integrate it. 

BR